RoganDawes/P4wnP1 #1: Cannot change language for duckencoder

ludoz, 2017-03-16:
Hello,

I tried to change the keyboard layout. So i change lang="us" by lang="fr" in the payload1.txt with no success.

So i commented the lang variable in payload1.txt and add it to setup.cfg. No more success ...

Nothing happen when i plug the Pi, when the lang is set to "us" it works !

So i decided to look deeper and when i start P4wnP1 with the script startup_p4wnp1.sh  and the lang defined to "us" , it's ok. The same test but with lang defined to "fr" give me that : 

`Traceback (most recent call last):
  File "/home/pi/P4wnP1/duckencoder/duckencoder.py", line 494, in <module>
    main(sys.argv[1:])
  File "/home/pi/P4wnP1/duckencoder/duckencoder.py", line 470, in main
    language=readResource(script_dir + "/resources/" + lang + ".properties")
  File "/home/pi/P4wnP1/duckencoder/duckencoder.py", line 23, in readResource
    val = splitted[1].strip()
IndexError: list index out of range`

Can you see where is the problem ?

mame82, 2017-03-16:
Yes...Problem resides [here](https://github.com/mame82/duckencoder.py/blob/6e2e68a5ca2ed07d2fe4d4e8bd3dff94ccc02981/resources/fr.properties#L201) The equal signs for the ISO chars are missing. I'm going to fix this as soon I have time to (next week). Thanks for testing "fr"

mame82, 2017-03-17:
@ludoz 
The issue should be fixed now. Could you please update the duckencoder submodule and run a test?

ludoz, 2017-03-18:
Thanks, it works for FR. I think you'll have the same problem with others language. When FR was not working i tested other azerty layout like de or es and they were not working. I say that for your information.

Thanks for this great platform. Lot of fun.

closed 2017-03-18 by ludoz

mame82, 2017-03-18:
I'll check the other language property files for similar bugs. Unfortunately I'm not able to test all layouts  to be working as intended on my own. You already helped with FR, thanks againg. DE and US have been tested by myself.