RoganDawes/P4wnP1 #5: hey i wanna ask if shd i be concerned

haxhxm, 2017-04-02:
arning: GMP or MPIR library not found; Not building Crypto.PublicKey._fastm                                                                                                             ath.
    building 'Crypto.Hash._MD2' extension
    arm-linux-gnueabihf-gcc -pthread -fwrapv -Wall -Wstrict-prototypes -fno-stri                                                                                                             ct-aliasing -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format                                                                                                             -security -fPIC -std=c99 -O3 -fomit-frame-pointer -Isrc/ -I/usr/include/python2.                                                                                                             7 -c src/MD2.c -o build/temp.linux-armv6l-2.7/src/MD2.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -W                                                                                                             l,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototype                                                                                                             s -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-securi                                                                                                             ty -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror                                                                                                             =format-security build/temp.linux-armv6l-2.7/src/MD2.o -o build/lib.linux-armv6l                                                                                                             -2.7/Crypto/Hash/_MD2.so
    building 'Crypto.Hash._MD4' extension
    arm-linux-gnueabihf-gcc -pthread -fwrapv -Wall -Wstrict-prototypes -fno-stri                                                                                                             ct-aliasing -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format                                                                                                             -security -fPIC -std=c99 -O3 -fomit-frame-pointer -Isrc/ -I/usr/include/python2.                                                                                                             7 -c src/MD4.c -o build/temp.linux-armv6l-2.7/src/MD4.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -W                                                                                                             l,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototype                                                                                                             s -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-securi                                                                                                             ty -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror                                                                                                             =format-security build/temp.linux-armv6l-2.7/src/MD4.o -o build/lib.linux-armv6l                                                                                                             -2.7/Crypto/Hash/_MD4.so
    building 'Crypto.Hash._SHA256' extension
    arm-linux-gnueabihf-gcc -pthread -fwrapv -Wall -Wstrict-prototypes -fno-stri                                                                                                             ct-aliasing -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format                                                                                                             -security -fPIC -std=c99 -O3 -fomit-frame-pointer -Isrc/ -I/usr/include/python2.                                                                                                             7 -c src/SHA256.c -o build/temp.linux-armv6l-2.7/src/SHA256.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -W                                                                                                             l,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototype                                                                                                             s -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-securi                                                                                                             ty -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror                                                                                                             =format-security build/temp.linux-armv6l-2.7/src/SHA256.o -o build/lib.linux-arm                                                                                                             v6l-2.7/Crypto/Hash/_SHA256.so
    building 'Crypto.Hash._SHA224' extension
    arm-linux-gnueabihf-gcc -pthread -fwrapv -Wall -Wstrict-prototypes -fno-stri                                                                                                             ct-aliasing -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format                                                                                                             -security -fPIC -std=c99 -O3 -fomit-frame-pointer -Isrc/ -I/usr/include/python2.                                                                                                             7 -c src/SHA224.c -o build/temp.linux-armv6l-2.7/src/SHA224.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -W                                                                                                             l,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototype                                                                                                             s -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-securi                                                                                                             ty -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror                                                                                                             =format-security build/temp.linux-armv6l-2.7/src/SHA224.o -o build/lib.linux-arm                                                                                                             v6l-2.7/Crypto/Hash/_SHA224.so
    building 'Crypto.Hash._SHA384' extension
    arm-linux-gnueabihf-gcc -pthread -fwrapv -Wall -Wstrict-prototypes -fno-stri                                                                                                             ct-aliasing -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format                                                                                                             -security -fPIC -std=c99 -O3 -fomit-frame-pointer -Isrc/ -I/usr/include/python2.                                                                                                             7 -c src/SHA384.c -o build/temp.linux-armv6l-2.7/src/SHA384.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -W                                                                                                             l,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototype                                                                                                             s -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-securi                                                                                                             ty -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror                                                                                                             =format-security build/temp.linux-armv6l-2.7/src/SHA384.o -o build/lib.linux-arm                                                                                                             v6l-2.7/Crypto/Hash/_SHA384.so
    building 'Crypto.Hash._SHA512' extension
[... cut ...]
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `dhcpcd' ov                                                                                                             errides LSB defaults (0 1 6).
insserv: warning: current start runlevel(s) (empty) of script `dnsmasq' override                                                                                                             s LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `dnsmasq' o

mame82, 2017-04-02:
There's no error. PyCrypto compilation succeeded (werror is a compiler argument).
The last messages don't make sense (runlevels aren't empty) but occurred on my tests, too. This isn't a problem, as long as the needed services are running. Test it by SSH'ing into the Pi

haxhxm, 2017-04-02:
Cant ssh tho but it works i have output as it shows 

```
Network is running
2 seconds delay before next output (delay test)
Delay passed by
P4wnP1 boot up finished
Next the output of command 'cat /tmp/Responder-Session.log' is sent to keyboard
Testing file output via keyboard
04/01/2017 11:17:21 AM - Responder Started: ['Responder.py', '-I', 'usb0', '-f', '-v', '-w', '-F']
```

mame82, 2017-04-02:
Seems your target box doesn't run a DHCP client on the network interface of P4wnP1.
Two choices here.
- enable DHCP client on target (assigned IP will be printed with keyboard messages)
- assign IP to rndis/ecm interface manually (IP 172.16.0.2, netmask 255.255.255.254)

Connect with ssh to `pi@172.16.0.1`

Note:
The master branch is a bit outdated, as I'm working in developed branch right now. Next release is able to work without Network at all (see [here](https://www.youtube.com/watch?v=MI8DFlKLHBk) )

After this feature is finished I'm going to rework the BTLM capture payload (not happy with it at the moment)

haxhxm, 2017-04-02:
u know someting bcs of you i wanna learn more python and help you i was thingking how do i get a bash bunny on pi zero then ur project came like a angel

mame82, 2017-04-02:
> u know someting bcs of you i wanna learn more python and help you i was thingking how do i get a bash bunny on pi zero then ur project came like a angel

Thanks, but most of the current code relies on bash to keep things simple. The new HID features rely on python and powershell. Python is a hood choice for rapid development in conjunction with networking tasks. Closing this issue now, as there's no bug. Feel free to contribute or develop with your fork. Developed branch will be merged into master soon, which changes some things!

closed 2017-04-02 by mame82