RoganDawes/P4wnP1 #6: Serial COM port connection

haxhxm, 2017-04-02:
is it possible to have a serial connection or can there be a new future to do that

mame82, 2017-04-02:
It is possible. In fact it was already implemented [here](https://github.com/mame82/P4wnP1/blob/master/startup_p4wnp1.sh#L139) and [here](https://github.com/mame82/P4wnP1/blob/master/startup_p4wnp1.sh#L186). 

I disabled serial support for two reasons:
- other USB functions have to be turned off to use it, because the UDC hasn't got enough EPs
- didn't felt the need of having it (considering the fact that not all possible gadgets could run in parallel)...device communication could be done via RNDIS / ECM or HID in future releases

haxhxm, 2017-04-03:
i mean i wanted have that bash bunny feel ><

haxhxm, 2017-04-03:
so how do i enable the settings ah just tak e the comment of is it

mame82, 2017-04-08:
Basically yes, but as said I've done no further testing, so I'm not able to tell if a TTY is already running on the serial interface. Please be aware of the fact that enabling ATM consumes additional USB EPs which could render other enabled gadgets useless, thus I'm not supporting it at the moment. I'm focused on finishing the HID covert channel to get a shell on P4wnP1. SSH could be used in conjunction with RNDIS/ECM, too. This has the benefit of sharing a internet connection to P4wnP1 which wouldn't be possible via serial.

closed 2017-04-19 by mame82