RoganDawes/P4wnP1 #7: Possibility of downloadable IMG

bytedeez, 2017-04-17:
Hi I am having tons of issues setting up rdnis as well as with wpa_supplicant via alternative method of gaining IC.

Anyway you could upload a downloadable img?
Thanks in advance.

mame82, 2017-04-17:
Unfortunately not possible at the moment. All Images I'm currently using include custom development code and aren't production ready.

I consider uploading a ready-to-go image, when devel branch is merged into master, but this takes time.

Sorry!
Maybe somebody else could help out. I'll leave the issue open for some days.

bytedeez, 2017-04-20:
Okay thank you. I'm sure I will eventually figure out my issues with setting up the pi zero as a USB dongle as well as the wpa_supplicant error. 
Might try an older version of raspbian from 03/02/2017. 
Correct me if I'm wrong but the kernel must be 4.4 or above right?

mame82, 2017-04-20:
Working Kernel is 4.4.50+

When Raspbian moves to the next Kernel, real issues could arise, but this isn't the case at the moment.

https://github.com/raspberrypi/linux/issues/1943#issuecomment-290928266

bytedeez, 2017-04-20:
Okay thanks you!
You are doing really awesome work here, can't wait til I can try it out for myself, until then keep it up!
I am going to close this for now.

closed 2017-04-20 by bytedeez