RoganDawes/P4wnP1 #10: Unable to install JTR

ZeteMKaa, 2017-06-09:
When running the install.sh script the following error occurs:

`Unpacking John the Ripper Jumbo edition...
tar (child): john-1-8-0-jumbo_raspbian_jessie_precompiled/john-1.8.0-jumbo-1_precompiled_raspbian_jessie.tar.xz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2`

The folder also seems to be empty when performing a git pull.
Let me know if you need more information

mame82, 2017-06-09:
I guess you haven't cloned the got modules (recursive mode).

See the following link for reference https://github.com/mame82/P4wnP1/blob/master/INSTALL.md

JtR is a got module which only gets cloned if you do it like this

`git clone --recursive https://github.com/mame82/P4wnP1`

ZeteMKaa, 2017-06-11:
My bad, that indeed did the trick. Thanks for your quick response.