RoganDawes/P4wnP1 #11: NULL pointer exception (master)

ZeteMKaa, 2017-06-09:
During start-up of default installation of the system the following error occurs:

`Jun  9 19:45:32 raspberrypi kernel: [   21.004694] Unable to handle kernel NULL pointer dereference at virtual address 00000002
Jun  9 19:45:32 raspberrypi kernel: [   21.010770] pgd = d8d00000
Jun  9 19:45:32 raspberrypi kernel: [   21.016721] [00000002] *pgd=18e89831, *pte=00000000, *ppte=00000000
Jun  9 19:45:32 raspberrypi kernel: [   21.022781] Internal error: Oops: 17 [#1] ARM
Jun  9 19:45:32 raspberrypi kernel: [   21.028640] Modules linked in: usb_f_hid usb_f_ecm usb_f_rndis u_ether bnep hci_uart btbcm bluetooth brcmfmac brcmutil snd_bcm2835 cfg80211 snd_pcm rfkill snd_timer snd dwc2 bcm2835_gpiomem uio_pdrv_genirq fixed uio libcomposite udc_core ipv6
Jun  9 19:45:32 raspberrypi kernel: [   21.041856] CPU: 0 PID: 626 Comm: bash Not tainted 4.9.24+ #993
Jun  9 19:45:32 raspberrypi kernel: [   21.048310] Hardware name: BCM2835
Jun  9 19:45:32 raspberrypi kernel: [   21.054815] task: da5f51c0 task.stack: d8dc2000
Jun  9 19:45:32 raspberrypi kernel: [   21.061438] PC is at alloc_ep_req+0x30/0xa8 [libcomposite]
Jun  9 19:45:32 raspberrypi kernel: [   21.068105] LR is at 0x0
Jun  9 19:45:32 raspberrypi kernel: [   21.074739] pc : [<bf090fe8>]    lr : [<00000000>]    psr: a0000013
Jun  9 19:45:32 raspberrypi kernel: [   21.074739] sp : d8dc3d98  ip : 00000000  fp : d8dc3db4
Jun  9 19:45:32 raspberrypi kernel: [   21.088058] r10: d8c761c8  r9 : da6370c0  r8 : d8c76170
Jun  9 19:45:32 raspberrypi kernel: [   21.094684] r7 : da6676a8  r6 : d8c3ff80  r5 : da50ef10  r4 : 00000008
Jun  9 19:45:32 raspberrypi kernel: [   21.101453] r3 : d8c3ffb4  r2 : 00000000  r1 : 00000000  r0 : d8c3ff80
Jun  9 19:45:32 raspberrypi kernel: [   21.108231] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jun  9 19:45:32 raspberrypi kernel: [   21.115011] Control: 00c5387d  Table: 18d00008  DAC: 00000055
Jun  9 19:45:32 raspberrypi kernel: [   21.121819] Process bash (pid: 626, stack limit = 0xd8dc2188)
Jun  9 19:45:32 raspberrypi kernel: [   21.128568] Stack: (0xd8dc3d98 to 0xd8dc4000)
Jun  9 19:45:32 raspberrypi kernel: [   21.135330] 3d80:                                                       00000002 d8c76170
Jun  9 19:45:32 raspberrypi kernel: [   21.142173] 3da0: 00000002 bf3e8c80 d8dc3dec d8dc3db8 bf3e77f0 bf090fc4 bf3d7994 bf3d7e8c
Jun  9 19:45:32 raspberrypi kernel: [   21.149059] 3dc0: 00000000 d8dc3dd0 bf08c37c d8c76170 da6676a8 da4d1de0 da66771c d8c76170
Jun  9 19:45:32 raspberrypi kernel: [   21.156004] 3de0: d8dc3e04 d8dc3df0 bf08c0c0 bf3e775c da6676a8 da4d1e8c d8dc3e3c d8dc3e08
Jun  9 19:45:32 raspberrypi kernel: [   21.162928] 3e00: bf090384 bf08c050 da5a6510 da4d1e54 d8dc3ea4 da68ae00 bf07bea4 da4d1de0
Jun  9 19:45:32 raspberrypi kernel: [   21.169984] 3e20: da4d1de0 bf07be98 00000000 da728000 d8dc3e5c d8dc3e40 bf079838 bf090144
Jun  9 19:45:32 raspberrypi kernel: [   21.177008] 3e40: 00000000 da68ae00 bf07bea4 d843f100 d8dc3e84 d8dc3e60 bf079d68 bf079808
Jun  9 19:45:32 raspberrypi kernel: [   21.184115] 3e60: 00000032 d843f100 d8d79540 da4d1c00 da4d1d90 0000000d d8dc3eac d8dc3e88
Jun  9 19:45:32 raspberrypi kernel: [   21.191257] 3e80: bf090d10 bf079c74 0000000d d8d79540 d8d79558 d8dc3f80 da6e08c0 01650c08
Jun  9 19:45:32 raspberrypi kernel: [   21.198396] 3ea0: d8dc3edc d8dc3eb0 c01baf30 bf090c6c d8dc3f80 da6e08c0 d8dc3f80 01650c08
Jun  9 19:45:32 raspberrypi kernel: [   21.205623] 3ec0: d8dc3f80 0000000d d8dc2000 00000000 d8dc3f4c d8dc3ee0 c0143dec c01bae68
Jun  9 19:45:32 raspberrypi kernel: [   21.212858] 3ee0: b6e98b90 d8dc3fb0 00000001 00000003 d8dc3fac d8dc3f00 c0009204 00000000
Jun  9 19:45:32 raspberrypi kernel: [   21.220174] 3f00: da4e3aa0 dac3f9c0 d8f1ae00 c01422c4 00000000 00000000 d8dc3f44 c0144be8
Jun  9 19:45:32 raspberrypi kernel: [   21.227523] 3f20: c01422c4 c0147320 d8dc3f4c 0000000d da6e08c0 01650c08 d8dc3f80 c000ffc4
Jun  9 19:45:32 raspberrypi kernel: [   21.234991] 3f40: d8dc3f7c d8dc3f50 c0144c24 c0143dc0 d8dc3f7c c0162040 da6e08c0 da6e08c0
Jun  9 19:45:32 raspberrypi kernel: [   21.242472] 3f60: 0000000d 01650c08 c000ffc4 d8dc2000 d8dc3fa4 d8dc3f80 c0145d34 c0144b80
Jun  9 19:45:32 raspberrypi kernel: [   21.250005] 3f80: 00000000 00000000 0000000d 01650c08 b6e98b40 00000004 00000000 d8dc3fa8
Jun  9 19:45:32 raspberrypi kernel: [   21.257636] 3fa0: c000fe40 c0145cec 0000000d 01650c08 00000001 01650c08 0000000d 00000000
Jun  9 19:45:32 raspberrypi kernel: [   21.265232] 3fc0: 0000000d 01650c08 b6e98b40 00000004 0000000d 01650c08 0000000d 00000000
Jun  9 19:45:32 raspberrypi kernel: [   21.272878] 3fe0: 00000000 bee542bc b6dc6008 b6e1d89c 60000010 00000001 00000000 00000000
Jun  9 19:45:32 raspberrypi kernel: [   21.280612] [<bf090fe8>] (alloc_ep_req [libcomposite]) from [<bf3e77f0>] (hidg_bind+0xa0/0x258 [usb_f_hid])
Jun  9 19:45:32 raspberrypi kernel: [   21.288552] [<bf3e77f0>] (hidg_bind [usb_f_hid]) from [<bf08c0c0>] (usb_add_function+0x7c/0x164 [libcomposite])
Jun  9 19:45:32 raspberrypi kernel: [   21.296544] [<bf08c0c0>] (usb_add_function [libcomposite]) from [<bf090384>] (configfs_composite_bind+0x24c/0x360 [libcomposite])
Jun  9 19:45:32 raspberrypi kernel: [   21.304808] [<bf090384>] (configfs_composite_bind [libcomposite]) from [<bf079838>] (udc_bind_to_driver+0x3c/0xe0 [udc_core])
Jun  9 19:45:32 raspberrypi kernel: [   21.313119] [<bf079838>] (udc_bind_to_driver [udc_core]) from [<bf079d68>] (usb_gadget_probe_driver+0x100/0x14c [udc_core])
Jun  9 19:45:32 raspberrypi kernel: [   21.321601] [<bf079d68>] (usb_gadget_probe_driver [udc_core]) from [<bf090d10>] (gadget_dev_desc_UDC_store+0xb0/0xcc [libcomposite])
Jun  9 19:45:32 raspberrypi kernel: [   21.330103] [<bf090d10>] (gadget_dev_desc_UDC_store [libcomposite]) from [<c01baf30>] (configfs_write_file+0xd4/0x17c)
Jun  9 19:45:32 raspberrypi kernel: [   21.338626] [<c01baf30>] (configfs_write_file) from [<c0143dec>] (__vfs_write+0x38/0x128)
Jun  9 19:45:32 raspberrypi kernel: [   21.347075] [<c0143dec>] (__vfs_write) from [<c0144c24>] (vfs_write+0xb0/0x1bc)
Jun  9 19:45:32 raspberrypi kernel: [   21.355553] [<c0144c24>] (vfs_write) from [<c0145d34>] (SyS_write+0x54/0xb0)
Jun  9 19:45:32 raspberrypi kernel: [   21.363967] [<c0145d34>] (SyS_write) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
Jun  9 19:45:32 raspberrypi kernel: [   21.372395] Code: ebff9c4d e2506000 0a000013 e5952024 (e1d230d2)
Jun  9 19:45:32 raspberrypi kernel: [   21.380928] ---[ end trace a87306446219c944 ]---
`

mame82, 2017-06-09:
This is a known, but unfortunately  unfortunately unaddressed issue in Kernel 4.9.x
There're multiple bugs in HID composite gadgets of this kernel version which render the USB gadgets stack useless, at least if HID devices should be used. This isn't caused by P4wnP1.

See here for details https://github.com/raspberrypi/linux/issues/1943

mame82, 2017-06-09:
Btw. The working kernel version used during P4wnP1 development is 4.4.50+

This version has its own (minor) issues with the RNDIS  composite module (described on the same raspbian thread). Hopefully both are gone in Kernel 4.11, for now kernel 4.4.50+ is the way to go

ZeteMKaa, 2017-06-12:
The downgrade for both the master as the devel branch worked perfectly! Use the following command to go to the 4.4.50 kernel:

`sudo apt install rpi-update
sudo rpi-update 52241088c1da59a359110d39c1875cda56496764`

Thanks for your help Marcus.

closed 2017-06-19 by mame82