RoganDawes/P4wnP1 #12: Error: "[!] Fingerprint failed"

4mby, 2017-06-21:
If I connect to Pi Zero's IP Address via PuTTY I always get the Error: "**[!] fingerprint failed**" directly after the "**[...] Poisoned answer sent to [..]**" message. I don't know what the Error is meaning and I don't know how to fix it. After a while the messages  are stopping and my normal terminal appears. If I check the Responder.db file (with sqlite3), it's empty. The Ducky HID-keyboard also doesn't do anything (maybe I'm just stupid).

Someone could help me, please?

ZeteMKaa, 2017-06-21:
The first error is because you have old ssh keys stored, you can remove these from the following registry key:
`HKEY_CURRENT_USER\Software\SimonTatham\PuTTY\SshHostKeys`

That the poising is done does not mean that the attack was actually successful. To test this you could access a non existing share on the Windows machine (\\example) to force authentication on the client. You will see a hash in Responder (screen session) if it works.

mame82, 2017-06-21:
@4mby

Although P4wnP1 is still under development (including major changes), let me try to help.

First I take some assumptions, which are missing in you description:

1) You're using the master branch, which hasn't  received any updates quit a while (same goes for devel branch for the last weeks as I'm really busy these days. Anyway, devel branch is the place where upcoming features get pushed in)
2) You're using `https://github.com/mame82/P4wnP1/blob/master/payloads/payload1.txt` which basically targets the issue described in [snagging creds from locked machines](machines) by Rob Fuller aka Mubix. So this early payload is somehow outdated, as the issue is well understood and was addressed with a patch. Anyway, there's an infamous software still prone to this issue, which I'm not allowed to mention here (till a patch is released). So there's still a chance to grab an NetNTLMv2 hash with this payload from Windows boxes.
3) You're targeting Windows  (using putty), although I don't know which version... according to the behaviour you described (fingerprint error) it should be Windows 10.

Your issues
>  "[!] fingerprint failed" 

Is an error (or information) created by responder, telling you not being able to fingerprint the OS of the target host. This happens for instance if you target Windows 10 and the network which is listened to by responder is declared as public (the default case for the mentioned payload). You can ignore this, because the purpose of this PoC payload isn't OS fingerprinting, but NTLM hash stealing. Anyway, the responder option for OS fingerprint is enabled in the payload script and mostly works on Win 7 boxes.

> [...] Poisoned answer sent to [..]"

Responder tries to spoof a name resolution request of your target (either LLMNR or NB) in order to redirect traffic destinated to this hostname to P4wnP1, which raises chances of grabbing hashes for this target host (for example by spoofing the host for an SMB request). According the assumptions above, this name poisoning attempts shouldn't succeed  (Windows 10 + P4wnP1 network declared as public)

According the poisoning problem and the fact that you haven't been able to fetch hashes (empty responder.db), refer to the recommendation given by @ZeteMKaa above.

> Ducky HID-keyboard

Needs to be clarified. But I suggest using the new keyboard payloads from devel branch. They use neat features like status LED of Pi, triggering after keyboard driver is loaded by target and triggering/controlling payloads from targets native keyboard LEDs (NUM, CAPS and SCROLL LOCK )

mame82, 2017-06-21:
@ZeteMKaa 

Thanks for supporting me in answering this issue. I'm really happy to receive some help to keep this project alive. 

Anyway, I guess the fingerprint error originates from responder, not SSH in this case

4mby, 2017-06-22:
@mame82 Thanks for the very detailed project. I think my questions are solved now. 
Thanks for this very nice project!

mame82, 2017-06-23:
The Video Shows an HID covert channel payload, which exists in devel branch.

A keyboard payload  starting different actions based on NUM / CAPS or SCROLL LOCK presses could be found here https://github.com/mame82/P4wnP1/blob/devel/payloads/hid_keyboard2.txt#L66

You have to create your own payloads or modify existing ones in order to different things.

I'm currently working on core features like implementing a HID covert channel based backdoor, which involves designing and testing multiple protocol layers and a client-server implementation. This consumes most of my time, thus I'm not providing payloads till the core functionality is finished.

Same goes for documentation: No docs till things are done.

According the payloads, they are easy to write:
- bash syntax
- callback functions could be used like shown in the demo payloads
- custom commands to control LED, print out via HID keyboard or set native keyboard triggers are available and could be used like shown in the example payloads

-There's no payload utilizing responder in the devel branch, right now. This doesn't mean you couldn't build one

Payload selection is done in setup.cfg

4mby, 2017-06-24:
@mame82 Thanks!

closed 2017-06-28 by mame82