RoganDawes/P4wnP1 #13: [devel]

ZeteMKaa, 2017-06-21:
Hi, during the boot process I see the following errors coming by:

Jun 12 02:22:53 raspberrypi bash[187]: File "/home/pi/P4wnP1/ledtool/ledtool.py", line 47, in <module>
Jun 12 02:22:53 raspberrypi bash[187]: File "/home/pi/P4wnP1/ledtool/ledtool.py", line 18, in prepare
Jun 12 02:22:53 raspberrypi bash[187]: with open(ledpath + "trigger", "w") as trigger:
Jun 12 02:22:53 raspberrypi bash[187]: IOError: [Errno 2] No such file or directory: '/sys/class/leds/led0/trigger'
Jun 12 02:22:53 raspberrypi kernel: [    0.000000] Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1680 bcm2708_fb.fbheight=1050 bcm2708.boardrev=0x9000c1 bcm2708.serial=0x776fb917 smsc95xx.macaddr=B8:27:EB:6F:B9:17 bcm2708_fb.fbswap=1 bcm2708.uart_clock=48000000 bcm2708.disk_led_gpio=47 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=5fb7e25a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes  rootwait
Jun 12 02:22:53 raspberrypi kernel: [    0.642522] leds-gpio: probe of soc:leds failed with error -2
Jun 12 02:22:53 raspberrypi kernel: [    0.642996] ledtrig-cpu: registered to indicate activity on CPUs

The location /sys/class/leds also seems to be empty from that point.

Uname: Linux raspberrypi 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux
Board: Raspberry Pi Zero W 1.1

mame82, 2017-06-21:
Weird `/sys/class/leds/led0/trigger` should always be present on raspbian sysfs to change trigger mode of LED.
The kernel errors stating that trigger got set back to "cpu activity" are confusing. 

I'm not able to reproduce this right now

mame82, 2017-06-21:
Could you comment out the following line of code
https://github.com/mame82/P4wnP1/blob/devel/boot/init_led.sh#L10

... and check if ...

`cat /sys/class/leds/led0/trigger`

.. is possible after reboot ?

If yes, please paste in the output

ZeteMKaa, 2017-06-21:
It fixes the error during the boot sequences. Only mention of led in the logs is:

```
Jun 12 03:41:44 raspberrypi kernel: [    0.642512] leds-gpio: probe of soc:leds failed with error -2
Jun 12 03:41:44 raspberrypi kernel: [    0.642990] ledtrig-cpu: registered to indicate activity on CPUs
```

But the directory stays empty

```
ls -la /sys/class/leds/
total 0
drwxr-xr-x  2 root root 0 Jun 12 03:55 .
drwxr-xr-x 51 root root 0 Jan  1  1970 ..
```

mame82, 2017-06-21:
As the P4wnP1 script is disabled now and there's still a problem with the LED kernel module propagating the interface files to sysfs, I guess this is some weird kernel problem (maybe caused by kernel upgrade/downgrade). I've no idea about the root cause, but the LED trigger file in sysfs should be present in a default raspbian installation.

Unfortunately I'm not able to test the devel branch on a clean raspbian installation, at the moment.
`/sys/class/leds/` should be populated (at least with trigger) on a clean raspbian install. P4wnP1 doesn't change this.

The default trigger for LED should be "mmc" (sdcard access), which is changed to "none" by P4wnP1 (to manually trigger the LED).

In your case the kernel seems to fail over to trigger the LED based on CPU usage, which is even more weird.

But a said... I'm not able to reproduce it (I'm working on two month old raspbian release but same kernel version).

ZeteMKaa, 2017-06-22:
My best guess is also the kernel downgrade. Saw ppl with the same issues that should have been resolved with an updated raspi-led module. Will try to test it on a fresh install if I have the time.

mame82, 2017-07-09:
Anything new on this? I'm not able to reproduce the issue

closed 2017-07-18 by mame82