RoganDawes/P4wnP1 #14: windows 10 Issue

Darkcast, 2017-06-30:
I recently created the p4wnp1 and pluged it into my windows 10 laptop and two interesting thing first I got this message http://bit.ly/2s97dAH (it could be part of the exploit but I don't know since I didn't see it in the instructions, the other one was that it could find the driver and I got an exclamation in the device manager. 
also, could I use this tool with your app ducktoolkit.com the reason I ask is because it create two files a .bin file and a payload and I dont know if bin is supported in your tool,

Thanks

mame82, 2017-07-09:
Hmm ... interesting, would be nice to know which payload triggered AV. Could you give more details ( used branch, used payload).

There should be no need to use ducktoolkit (not my app), as the devel branch supports pure duckyscript. See the hid_keyboard payloads for usage examples: https://github.com/mame82/P4wnP1/tree/devel/payloads

mame82, 2017-07-10:
Reviewing the warning dialog about `wmiav.exe`, I doubt this is caused by a P4wnP1 payload. The first Impression was that one of my payloads triggered some Kaspersky heuristics, but this wouldn't end in a Windows dialog asking for trust of an AV component. So no, this isn't pat of any of my payloads!

closed 2017-07-18 by mame82