RoganDawes/P4wnP1 #15: Usb device not recognized + no internet connection

locomain, 2017-07-10:
After running the installer Windows 10 notifies me that the usb device is not recognized and i am unable to connect via putty.
I tried the master and devel branch on a pi zero with a clean raspbian lite.

gautaus, 2017-07-12:
Same here ! Device Descriptor Request failed

mame82, 2017-07-12:
Please give some more details:
Kernel Version of raspbian, used payload, output of Pi (hdmi) and about the unrecognized device in device manager.

Maybe I'm able to help

gautaus, 2017-07-12:
It's 
Linux raspberrypi 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux
I am unable to see hdmi output because I don't have a screen. I ssh with the pi's IP address.

gautaus, 2017-07-12:
SSHed via wifi and got this output : 

Waiting for responder to be started before attaching to screen session
Press <CTRL> + <C> to abort waiting
...............................................................................................................................................

This goes on forever

locomain, 2017-07-13:
[image]

hope this helps

mame82, 2017-07-13:
Unfortunately the current raspbian release ships with a new kernel Version (since 05-07-2017).
P4wnP1 works with kernel 4.4.50+.

The 4.9.x kernels have known Problems as discussed [here](https://github.com/mame82/P4wnP1/issues/11) and [here](https://github.com/raspberrypi/linux/issues/1943).

A kernel patch meant to solve the USB Problems was pushed into raspbain kernel branch yesterday. Anyway, testing it needs time... till the issues on raspbian are fixed, the way to go is kernel 4.4.50+.

Btw: The old 4.9.x kernels crashed with "kernel panic", which obviously isn't true in your case. Thus there are new problems arising with the USB gadget modules, which are needed for P4wnP1 to work.

gautaus, 2017-07-13:
Thanks a lot ! I will try with an older image.

gautaus, 2017-07-13:
Its already running the payloads and not allowing me to do anything. This is the ssh via wifi.

SSH 172.16.0.1 on windows box putty times out. 

[+] Poisoners:
    LLMNR                      [ON]
    NBT-NS                     [ON]
    DNS/MDNS                   [ON]

[+] Servers:
    HTTP server                [ON]
    HTTPS server               [ON]
    WPAD proxy                 [ON]
    Auth proxy                 [OFF]
    SMB server                 [ON]
    Kerberos server            [ON]
    SQL server                 [ON]
    FTP server                 [ON]
    IMAP server                [ON]
    POP3 server                [ON]
    SMTP server                [ON]
    DNS server                 [ON]
    LDAP server                [ON]

[+] HTTP Options:
    Always serving EXE         [OFF]
    Serving EXE                [OFF]
    Serving HTML               [ON]
    Upstream Proxy             [OFF]

[+] Poisoning Options:
    Analyze Mode               [OFF]
    Force WPAD auth            [ON]
    Force Basic Auth           [OFF]
    Force LM downgrade         [OFF]
    Fingerprint hosts          [ON]

[+] Generic Options:
    Responder NIC              [usb0]
    Responder IP               [172.16.0.1]
    Challenge set              [random]
    Don't Respond To Names     ['ISATAP']
....
....
....
....
[!]  Fingerprint failed
 [!]  Fingerprint failed
 [!]  Fingerprint failed
[!]  Fingerprint failed
[*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name wpad
 [*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name kepjqvdebzdv
[*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name pjzphldatwwn
 [*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name afrdpyfnagyqfv
[!]  Fingerprint failed
[*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name wpad
[!]  Fingerprint failed
[*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name Asphalt
[!]  Fingerprint failed
 [!]  Fingerprint failed
[!]  Fingerprint failed
[*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name hjbtbjfpwz
[*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name nbovzauusvnlaf
[*] [LLMNR]  Poisoned answer sent to 169.254.254.187 for name hadgclycytnxcwb
[!]  Fingerprint failed

How do I control payloads ? Am I missing something here. This is an old kernel btw.

Linux p4wnp1 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux

closed 2017-07-18 by mame82