RoganDawes/P4wnP1 #16: Feature Request

beezTEM, 2017-07-27:
I love this piece of software, but wish there was an easy way to run something automatically, after stage 1 on the backdoor was ran, I want to run a powershell, download a file to C:\Windows and then exit, allowing for entering into the cmd line.

The file I want to download is here: https://beeztem.github.io/downloads/nir.exe (selfhosted file of http://www.nirsoft.net/utils/nircmd.html , just the exe though)

I am downloading to C:\Windows for easy use from the command line.

If anyone wants to help me with this, that would be great.

Thank you for your time and consideration.

mame82, 2017-07-27:
> was an easy way to run something automatically, after stage 1 on the backdoor was ran

If I got it right, you ask for something like a "RunScriptOnClientConnect" command.
So yes, this should be implemented as feature. But if I do this, I'm going to restrict command execution to "P4wnP1 shell" internal commands.

>  I want to run a powershell, download a file to C:\Windows and then exit

There are tons of ways to do "Download&Eexecute" but I'm not willing to build this into P4wnP1 (it is too loud, at least till I implemented file transfer into memory). If you use google, you'll find multiple solutions to achieve this task, in fact it is possible with P4wnP1 in a one liner (using cmd.exe to chain commands). Example:

```
P4wnP1 shell (client not connected) > FireStage1
Target connected through HID covert channel

P4wnP1 shell (client connected) > CreateProc cmd.exe /c echo command1 & echo command2 & pause
P4wnP1 shell (client connected) > Process with ID 10824 created

P4wnP1 shell (client connected) > interact 10824
Trying to interact with process ID 10824 ...
command1
command2
Dr▒cken Sie eine beliebige Taste . . .

Proc with id 10824 exited
P4wnP1 shell (client connected) >
Process exited... stopping interaction
P4wnP1 shell (client connected) > KillClient
P4wnP1 shell (client connected) > Client received terminates!
P4wnP1 shell (client connected) >
Target disconnected
P4wnP1 shell (client not connected) >
```

The example chains `echo command1` and `echo command2`. The `pause` is there to keep the process active for interaction, otherwise it would kill itself after execution of the 2 commands.

So if I implement "RunScriptOnClientConnect" your autorrun script would look something like this

```
CreateProc cmd.exe /c [command1] & [command2] ... & [command n]
KillClient
```

Of course comand1 ... commandn have to be replaced by your download and execute command chain.
 
Deploying the autorun mechanism could automate this.

Did I get it right?

beezTEM, 2017-07-27:
Yes! that is really close to what I was looking for, the file transfer would be really nice so that it wasn't so loud and just stayed in the network! Thank you so much for the help.

closed 2017-07-27 by beezTEM