RoganDawes/P4wnP1 #17: NotAnIssue

4mby, 2017-07-27:
I just wanted to thank you! This is a very very good work and it works perfectly! I just thought of making a video about it, but hopefully seytonic or someone else make a video about this so this becomes famous. 
Thanks to the developer!

mame82, 2017-07-27:
@4mby Thanks for your motivating feedback. 
 
P4wnP1 is developed as platform giving the user a basic set of capabilities (it doesn't ship with many payloads on its own).  So finding creative ways to use the tool is up to the community and I'm happy about every feedback, no matter if it is in form of a text or a video. I could imagine, that it could be used in ways I didn't even think about during development. Looking forward to see guys like you working with it.

mame82, 2017-07-28:
> hopefully seytonic or someone else make a video about this 

He did ... https://www.youtube.com/watch?v=Pft7voW5ui8

4mby, 2017-07-28:
Nice!

closed 2017-07-29 by mame82

DRSDavidSoft, 2017-07-30:
Hey @mame82, and nice work.

Could you explain the exploits in the repo? I came across Stage1.dll, but unlike the others, there doesn't seem to be any info on it.

Thanks for your work!

mame82, 2017-07-31:
@DRSDavidSoft Stage1.dll is part of "hid_backdoor" payload, which is explained in README.txt