RoganDawes/P4wnP1 #18: Great.... Running into issues

beezTEM, 2017-07-27:
I went to plug this into my friends computer and hide it under his desk, and discovered, that no matter what USB cable I connected, it would boot into interactive mode, this is very annoying but I was wondering, is there a way to disable interactive mode? If not, is there some other way to make this work?

My friends computer is a Dell All-In-One from a year I can't remember.

I tried 3 USB cables, my Sony PS4 Controller charging cable (this worked on my laptop), a generic USB cable (also worked), and another generic USB cable (didn't work on laptop).

Any help would be great! Thank you in advance.

mame82, 2017-07-28:
Sounds weird. Anyway, i like questions containing things like "... a friend of mine ..." or "... somebody I know ...".

So yes, interactive mode detection could be disable.

Change the line
`detect_usb_hostmode # creates OTG_MODE=true if P4wnP1 is in OTG mode`
into 
`OTG_MODE=false`
in the following file
https://github.com/mame82/P4wnP1/blob/master/boot/boot_P4wnP1#L60
and you should be fine.

mame82, 2017-07-29:
Have you been able to get it running with the hint given above

mame82, 2017-07-29:
@beezTEM could it be, that you friend sold its dell meanwhile? 
Anything new on this?

mame82, 2017-07-31:
Going to close this issue

closed 2017-07-31 by mame82

beezTEM, 2017-09-20:
Sorry about not responding, I did get it to work by disabling the check