RoganDawes/P4wnP1 #20: Does it work with the chip

rogafe, 2017-07-27:
Does it work with [this](https://getchip.com/pages/chip) ?

mame82, 2017-07-28:
I doubt it does, see `FAQ.md` for reference.

rogafe, 2017-07-28:
Thank I test when I come home I will tell you

whackyhack, 2017-07-28:
Yes it works.  Use CHIP OS 4.3, load g_ether. (Unload g_serial.)

mame82, 2017-07-28:
@rogafe thanks

mame82, 2017-07-28:
@whackyhack could you confirm that OTG mode and LED controlling works as intended

rogafe, 2017-07-29:
There is problem that i can't resolve now for activate the usb hid we need to flash the chip with an USB-UART and i dont have one ... 
I will buy one when I come back from holidays. Do I have to close the issue ?   

PS : Sorry for my English i am not an native speaker.

mame82, 2017-07-29:
> Do I have to close the issue ?

Why should you? Let's see where this gets!

> Sorry for my English i am not an native speaker.

Same goes for me, not sure if I'm native speaker in my own language, at all!

> If you know an other technique to flash the chip without the usb-uart let me know .

Unfortunately not, I stay away from micro controller world whenever possible. But take a look at USaBUSe, many of the problems you are going to face have been solved by Rogan Dawes, I guess.

mame82, 2017-07-29:
Hmm the last a answer was a bit misplaced, as you're not talking about a micro controller solution. So there is a USB-serial solution, but I doubt it is helpful: you could enable USB serial gadget on a Pi zero and forward the data to its serial connectors, I think. But as said, this isn't helpful in your case

rogafe, 2017-07-29:
@whackyhack how did you do this ?

mame82, 2017-07-29:
Never done it, but here's at tutorial on how to use the gpio pins for serial communications 
http://pwiatrowski.com/technology/raspberry-pi-zero-esp8266-internet/

An example of how to enable USB CDC_ACM gadget (USB serial ) was in my early code
https://github.com/mame82/P4wnP1/blob/894a81ac3893ab77879ada4fca50732fcd390ac2/startup_p4wnp1.sh#L139

So I guess these two serial ports have to be interconnected, maybe slattach could do this job. 

But you're on your own with this.

rogafe, 2017-07-29:
Ok, I try to install and the script say i don't have raspbian so i comment the install and run it now I pray and it install  
[image]

rogafe, 2017-07-29:
Ok I run with some issues
First no wifi running 
the ship don't come with the pi users so i create it

rogafe, 2017-07-29:
So it dont run the wifi because  the chip dont have iwconfig so i add it to the install

mame82, 2017-07-29:
Wifi capability detection is based on this
https://github.com/mame82/P4wnP1/blob/master/wifi/check_wifi.sh

if wlan0 isn't shown by iwconfig, this gets false.

This again means that install.sh doesn't install hostapd and the boot scripts don't try to enable the hostapd based AP.

If the wifichipset is supported by hostapd , you maybe have to change the driver in hostapd config.

Here's the file responsible for WiFi setup (dhcp + hostapd)
https://github.com/mame82/P4wnP1/blob/master/boot/init_wifi.sh

rogafe, 2017-07-29:
Thank,I am a bash beginner but i will try to make it work but as i say before monday i will go in vacation so my work will be slow or may be stop for 1-2 weeks

whackyhack, 2017-07-29:
I haven't tried HID as my main interest is in network. (I realise that HID is the main focus here.)  People have experimented with USB HID on CHIP.  Maybe you can ask in CHIP community.

mame82, 2017-07-29:
Most important is, that you get the composite gadget with HID running, and that this whole thing could be configured via filesystem.

rogafe, 2017-07-29:
but how i can test the composite HID ? when i log into the chip with the serial i dont have the P4wnPi shell

mame82, 2017-07-29:
First check if device files `/dev/hidg*` are created. You could start the sever by hand: `python hidtools/backdoor/P4wnP1.py`.

The HID device files need to be accessed with root privs, so start the server from root user

rogafe, 2017-07-29:
So, I dont have the usb hid enlable

rogafe, 2017-07-29:
And we go back to the need of flash an new kernel things i can't do because a dont have the good cable

rogafe, 2017-07-29:
Ok i fund an tutorial where there explain how to use an arduno as an ttl converter - but i dont have my jumper cable so ... ( that it the bad thing about change home )

mame82, 2017-08-06:
@rogafe anything new on this?

whackyhack, 2017-08-09:
I found this tutorial about loading g_hid on libcomposite by renzo: http://www.raspibo.org/wiki/index.php/Chip:_USB_hid_on_libcomposite.  I still haven't tried but this means the technical path is open. (It requires a custom 4.3 kernel, but does not require messing with SDK, uboot, etc. if you are already on 4.3.  He actually wrote a howto about compiling your own: http://www.raspibo.org/wiki/index.php/Compile_the_Linux_kernel_for_Chip:_my_personal_HOWTO)

I'm still trying to work around oddities in 4.3's networking, the only version that includes RDNIS support.

rogafe, 2017-08-09:
Now like I say I am in vacation, I go home the 13 and after that I will be on a family wedding so that part of the project might be a bit paused. But if I can make my old laptop work I can work on my family house so I keep in touch. Sorry for the lack of new

rogafe, 2017-08-09:
And thanks @way ( sorry no complet name on android...) anyway I will look at it

rogafe, 2017-08-16:
@whackyhack did i need an usb ttl or something like that ?

rogafe, 2017-08-17:
So, i have found my usb - ttl cable so i will be testing soon 

ps : i have buy an raspberry pi zero for testing to make an comparison between the two ( and maybe a can do something with the (pocket)chip and the raspberry like an gui style on the pocket chip or something like that

rogafe, 2017-08-19:
UPDATE 
I have some error with my usb-ttl  on windows 10 ( maybe because is not a official one ) 
I have to choice now install an linux ( make space on my hard drive) or try to make it work 

I chose the first one and i will come back to you when i make it work

rogafe, 2017-08-21:
TL;DR I made the usb ttl work

the label on the device is not the good :(

rogafe, 2017-08-22:
@mame82 your script need /dev/hidg1  and the way i found to create a hid device on it create /dev/hidg0
where did you defind this 
I also tried to get the wifi work nothing appear on my phone ( i try also with an usb key on my computer )

rogafe, 2017-08-22:
So with your init_usb.sh i can activate hidg0 but when i try to use 
python duckencoder/duckencoder.py -i DuckyScripts/HelloWorld.duck -p | python hidtools/transhid.py > /dev/hidg0 it dont work

mame82, 2017-08-23:
@rogafe 
I bring up two HID devices (at least for backdoor and frontdoor payload).
The crucial parts are the correct report descriptors.
The keyboard descriptor is basically the default one from YSB usage pages.
The second one, for the raw HID device is handcrafted (created with descriptor tool referenced in USB specs).

The second device gets initialized here:

https://github.com/mame82/P4wnP1/blob/master/boot/init_usb.sh#L90

The naming of the final devices is done vy Raspbian (from my ovservations in order of initialization).
So I end up with hidg0 for keyboard and hidg1 for my generic HID device

mame82, 2017-08-23:
transhid.py is a small tool to convert encoded ducky script to my own HID keyboard report format

mame82, 2017-09-18:
Is this still alive?

closed 2017-09-26 by mame82