RoganDawes/P4wnP1 #21: Pi Zero Wifi Over ESP8266?

bfayers, 2017-07-28:
There is this: http://pwiatrowski.com/technology/raspberry-pi-zero-esp8266-internet/ guide for wifi on pi0 with an esp8266 pimoroni also sell [an esp8266 phat](https://shop.pimoroni.com/products/esp8266-phat)

It would be slow (1Mbps) but surely, thats enough for ssh to be used with the HID backdoor?

Just a small suggestion..

mame82, 2017-07-28:
Hard to say if this would work. Not sure whats the advantage of the "serial to WiFi mod" compared to a Pi Zero W. Anyway the HID channel is tied to 64000 Bytes per seconds. My first protocol layer produces additional overhead reducing the rate to 62KBytes per second. The other layers reduce speed even more (depending on size of the data streams transfered and how they could be fragmented).

So yes, 1 Mbps is enough to relay the HID channel

mame82, 2017-07-29:
I forgot to mention the most important thing... Everything you're aiming at has been done, in fact an ESP8266 was used without a Pi ;-)

https://github.com/sensepost/USaBUSe

But if you've read the credits, you already know.

mame82, 2017-07-31:
@bfayers are you still on this? If not I'm going to close this issue

mame82, 2017-08-02:
Closed due to missing feedback, could be reopened if needed.

closed 2017-08-02 by mame82