RoganDawes/P4wnP1 #23: Alternative

harshil2105, 2017-07-29:
Is there no alternative other then raspberry pi zero/Raspberry pi W because standalone shipping of this product costs around 30$ then others taxes..could this be used in orange pi or rpi3?

whackyhack, 2017-07-29:
Not sure about Orange Pi but RPi3 is out of question.  Zero is the only Pi line that comes with OTG.

harshil2105, 2017-07-29:
I'm not sure....you are a researcher so maybe I thought you could tell us

<https://github.com/mame82/P4wnP1/issues/23#issuecomment-318807900>, or mute
the thread

whackyhack, 2017-07-29:
Indeed, Orange Pi (One and Plus) would work - provided you supply power to it.  It has a rare hardware combination, as its USB OTG does not provide power.  Orange also released a Zero that allows power supply via OTG so that would be a better candidate.

harshil2105, 2017-07-29:
So you mean a orange pi zero would work in our case....please check this
http://www.orangepi.org/orangepizero/

mame82, 2017-07-29:
Hey guys, Orange Pi hasn't been tested. Please read `FAQ.md` where I explain several conditions which tie P4wnP1 to a raspbian based Raspberry Pi Zero. Especially the access to the UDC's DebugFS interface could lead to problems.

Anyway, would be happy to see this running on another board

harshil2105, 2017-07-29:
Bro do you sell raspberry pi zero they cost a lot in shipping maybe you
could offer a solition because they cost like 30$ for a raspberry pi zero

harshil2105, 2017-07-29:
Maybe your community might help us in testing it on orange pi

mame82, 2017-07-29:
I hope they do, because here a Pi Zero W costs 11 Euro and was a cheap solution with large customer base. In fact I haven't been aware of the fact that they are this pricy elswhere.

Anyway, its up to you guys to port to different hardware. As far as I'm able I give support. But as this is a spare time project (stated in the readme) I'm a bit restricted and I'm not willing to biy every device on market ro test on. This is one of the reasons for making the project open source.

harshil2105, 2017-07-29:
Are you interested in shipping raspberry pi zero? You could make a lot of
profit and it would support your opensource project....you could buy more
boards to test on and it will financially help you as well

mame82, 2017-07-29:
Should other people earn the profit, I'm fine, but yet a bit suprised how hard it is to get a Pi Zero W.

harshil2105, 2017-07-30:
Can you just send it to me I'm a fan of your work and really want to test
it...I even can't get a rpi zero w..I will pay

qwerty123443, 2017-07-30:
@harshil2105 what country are you in?

harshil2105, 2017-07-30:
Right now I'm india...I have a meeting over here and I won't be here a long
time...ill be here for like a month at the max...I travel a lot

mame82, 2017-07-31:
I'm going to close this! If there're further questions, please open a new issue

closed 2017-07-31 by mame82

harshil2105, 2019-03-05:
it kinda works on orange pi zero