RoganDawes/P4wnP1 #24: Im really confused on how I would get this to run a ducky script when plugged in.

seiterseiter, 2017-07-29:
Could you write out some simple instructions or something so I know how I can get it to run a ducky script when plugged in.

Thanks :)

mame82, 2017-07-29:
`hid_keyboard.txt` is as simple example of using inline DuckyScript in a payload.
`hid_keyboard2.txt` is a more sophisticated example, branching payload execution based on keyboard interaction from the target host.

`cat $wdir/DuckyScripts/HelloWorld.duck | outduck` placed into the onKeyboardUp callback of a new payload, would run this script on boot.

Using the backdoor payload the `SendDuckyScript` command allows to run scripts on demand.

So there multiple ways. It is left to the community to build payloads as needed.
It should be noted, that using P4wnP1 only to boot and run a keyboard script is a bit oversized, as this is not its main purpose. If this is your use case, I suggest buying a 1$ DigiSpark and use my Duck2Spark project, because it will boot much faster.
P4wnP1 is meant to combine USB keyboard attacks with other USB attacks, backed by a full fledged Linux system, in order to build complex chained attacks.

Hope this helps

seiterseiter, 2017-07-29:
The issue with a DigiSpark is that it doesnt allow the use of an sd card..

bfayers, 2017-07-29:
@seiterseiter This isn't designed to run on a DigiSpark

DigiSpark's are microcontrollers not SBCs

You want this: https://www.youtube.com/watch?v=fGmGBa-4cYQ by seytonic

seiterseiter, 2017-07-29:
@bfayers Im aware of that, im using a raspberry pi zero, I was just responding to @mame82 about using a DigiSpark and it not having an sd card.

mame82, 2017-07-29:
@seiterseiter I see, seems I have to put some effort into Mass Storage support. Fo now UMS mode is backed by a 128 MB image.

If I add in a drive laybel and allow mounting the image into P4wnP1 FS when USB mass storage is unused, it could nicely be incorporated into DuckyScript based attacks. It would even be possible to emulate an optical drive and mount ISOs which could be used from KeyBoard payloads.

But this is a big ToDo

bfayers, 2017-07-29:
@seiterseiter My bad.

mame82, 2017-07-29:
One more idea I should mention, in order to help to think it the "P4wnP1 way". If you want to exfiltrate data or do things like download&execute and your initial vector is Keyboard, you could build payloads like this.

Install some kind of webserver on P4wnP1 (apt is there for you).
Create a payload with RNDIS or CDC_ECM and HID keyboard enabled. Use the onNetworkUp or onTargetGotIP callback to to be sure the target has connected, and run `cat $wdir/DuckyScripts/your_download_exec_script.duck | duckhid` from there.

In the duckyscript place a RubberDucky payload, which downloads and execute from P4wnP1 by accessing http://172.16.0.1/path/to/your/evil/file.

Uploading could be done the same way, for instance if you use an FTP sever or setup the webserver to handle WEBDAV (while your duckyscript should use some webclient capable of sending PUT request )

What I try to explain is: You could achieve much more if you use the possibilities given, not only keyboard output.

battletux, 2017-08-01:
@seiterseiter This is how I have been testing P4wn-P1 and running the included duckyscript examples:

First of all install P4wn-P1 (duh!) I did this via Method 2 detailed in the INSTALL.md doc with an RPi3 which installed git and P4wn-P1, after first running
`sudo apt update; sudo apt upgrade -y`.
Then I power off the RPi3 with `sudo poweroff`.

Next up insert the sd card into the RPi0w boot up using the OTG cable in the USB port on your PC and ssh to the default IP (172.24.0.1) with username 'pi' and password 'raspberry'. Configure it to use the HID covert backdoor, by uncommenting the "hid_backdoor.txt" payload in the payloads section of  ~/P4wnP1/setup.cfg.

Now reboot and connect it to a target machine. Give it a few minutes to boot up and setup the WiFi hotspot fully then connect to the P4wnP1 hotspot and ssh to the RPi0w (same IP and username as before).

Once connected run `SetKeyboardLanguage` and enter your language, failure to do this will cause the initial exploit to fail and a load of stuff to be entered into the powershell window.

_Make sure you set the language to that of the victim's machine._

Once the Keyboard language is set run the initial exploit: `FireStage1`. Now when running this I could not get this to work unless the machine was unlocked prior, so make sure you have someone distract the user so they do not see the 2 brief pop-ups.

Once done and you are connected to the client you can run duckyscripts that are located in the ~/P4wnP1/DuckyScripts folder by running `SendDuckyScript` and selecting the script to run.

I may be going about this the long way but it works for me in my limited testing so far.

mame82, 2017-08-01:
Souns like you have it working.
`FireStage1` is only needed to bring up the covert chanbel, in order to receive output of remote processes (for example `shell` command).

`SendDuckyScript` works withou connecting this channel and thus withou `FireStage1`

closed 2017-08-02 by mame82

ChopsKingsland, 2019-05-28:
@mame82 mine won't even boot into the shell!