RoganDawes/P4wnP1 #25: There is no screen to be detached. - hid_backdoor

chris1000, 2017-07-30:
Let me start off by saying that this project looks awesome! Great job Mame82!

I just did a fresh install and am able to connect to the network that the Pi creates but every time I SSH into the Pi waiting for the remote attack shell to open I just get a message saying "There is no screen to be detached." and I can't figure out why. I've run the install script with sudo and without sudo to see if that fixes the problem but I haven't had any luck. I also uncommented the write line and recommented it just to see if that makes a difference. 

Any help would be great!

(I am using the Pi 0 W and it is updated to the latest version of Raspbian lite)

mame82, 2017-07-30:
Thanks for your compliments.

Considering your issue (I guess you use the backdoor payload):

There's a corner case, because the server for the HID channel on P4wnP1 is started from the onKeyboardUp function. If the target doesn't manage to load the keyboard drivers fast enough (or load them at all) for some reason, this server will not be started. To have the server interface available on login, it is attached to a screen. If the screen isn't running in login, you receive this error.

So be sure your target has been done loading the keyboard driver, before login via SSH. You could logout an login again with SSH if this happens, until the server is loaded or run 'sudo screen -d -r' to try to reattach the screen again

mame82, 2017-07-31:
Do you still have problems with this?

chris1000, 2017-08-01:
Unfortunately yes, I've even completely reinstalled Raspbian but no luck.

This message was automatically generated by Gmail.

mame82, 2017-08-01:
Could you please add some infos:

- What's your target OS.
- Has the target managed to install the keyboard drivers?
- What is the output if your run `sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py` by hand ?

Additionally, please check if there's content in `/home/pi/P4wnP1/john` to assure you cloned recursively.

Floating-Feline, 2017-08-01:
Hello @mame82 I am also a huge fan of the work your doing here and am also having the same problem as chris. The target machine on my end is a windows 10 pro box 64x, version 1703. The drivers for the keyboard did install and when you run the backdoor payload by hand it gets served up, but unfortunately the client wont connect. If theres anything else you could suggest or want me to try feel free, ill be on the pc all day. Thanks!

[image]

mrhut10, 2017-08-01:
I have the same issue.
installed all no problem, then when plug in to any of the win10 PC's i have in my home.
and log in, it never goes into the menu screen that it is meant to go into.

doesn't seem to matter how long i wait either.

mrhut10, 2017-08-02:
sorry resolved, turns out all my issues came from not having a good USB cable.

might be a good idea to verify everyone has a working data usb cable,
it fixed the same issue everyone had above for me.

mame82, 2017-08-02:
@Floating-Feline 
Please exit the server by issuing the `exit` command and provide the output of:
`ls -R /home/pi/P4wnP1/duckencoder/`
additionally the output of
`ls /dev/hidg*`

mame82, 2017-08-02:
@mrhut10 good to hear

mame82, 2017-08-02:
@chris1000 is your problem solved?

mrhut10, 2017-08-02:
@mame82 awesome project btw..

-------- Original Message --------

Swiftb0y, 2017-08-02:
Just to give my two cents...
This is also the case when exiting from the shell and reconnecting via ssh afterwards.
So the shell would have to be started when connecting via ssh as well as when the drivers were loaded. Otherwise I have to reboot the PI. (which also makes sense)

mame82, 2017-08-02:
@Swiftb0y this isn't an issue, but a lack of documentation.
`exit` ends the backdoor server, that's the purpose of that command.
On boot the server is attached to a screen, which is then attached to a users session on login (as long as the server is running).

If you want to detach the server screen, without exiting the server, use `ctrl+a` followed by `d` (see screen docs for reference.

If you want to reattach to the screen holding the server, run `screen -r`.

If you do multiple ssh logins, the server screen will always be detached from the former SSH session and reattached to the new one, this is how it's meant to be played!

Swiftb0y, 2017-08-02:
Ahhh thanks. I just thought that might help with resolving the issue.

battletux, 2017-08-04:
I seem to be getting this behaviour when connecting to OS X Sierra. If I then plug it into a windows 7 machine I get the normal server interface.

If I have sometime today (unlikely) I'll have a play and see why this happens.

mame82, 2017-08-06:
@battletux the backdoor payload is dedicated to Windows targets. Anyway, keyboard commands like `SendKeys` or `SendDuckyScript` should work against OSX, too. So let's see where you get with testing, maybe we could fix it.

mame82, 2017-08-09:
@chris1000 I'm closing this. Please reopen the issue if you still are facing problems

closed 2017-08-09 by mame82

Amon-Amarth, 2017-08-09:
@chris1000 Same thing happened to me, but i fixed it by re-creating the symbolic link in (using ln -s): P4wnP1/hidtools/backdoor/DuckEncoder.py ->../../duckencoder/duckencoder.py
P4wnP1/hidtools/backdoor/resources -> ../../duckencoder/resources/

Also for some reason when i downloaded the whole project from the website to my computer and added the file to the Pi, duckencoder folder was empty. i had to manually download everything and SCP to the Pi. 

Once all of this is done, i sudo python P4wnP1.py and was able to get the menu. 

Hope this helps.

BTW mame82, maybe i should open another thread for this, but when first installing the project hostpad was NOT installed. Looking at the install.sh I've found this IF statement below : 

if $WIFI; then
# install dhcpd, git, screen, pip
	sudo apt-get install -y dnsmasq git python-pip python-dev screen sqlite3 inotify-tools hostapd
else
	sudo apt-get install -y dnsmasq git python-pip python-dev screen sqlite3 inotify-tools
fi

not sure what happened here, but i believe that $WIFI return false during the installation script....

mkbcaptain, 2017-08-23:
There is no screen to be detatched error here.

Output of 'sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py'
Traceback (most recent call last):
  File "/home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py", line 855, in <module>
    HIDin_file = open(dev_file_in_path, "rb")
IOError: [Errno 2] No such file or directory: '/dev/hidg1'
Cleaning Up...
Traceback (most recent call last):
  File "/home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py", line 891, in <module>
    ll.stop() # send stop event to read and write loop of link layer
NameError: name 'll' is not defined

John folder is full of stuff

Output of 'ls -R /home/pi/P4wnP1/duckencoder/'
/home/pi/P4wnP1/duckencoder/:
duckencoder.py  README.md  resources

/home/pi/P4wnP1/duckencoder/resources:
be.properties  de.properties  gb.properties        pt.properties  us.properties
br.properties  dk.properties  hr.properties        ru.properties
ca.properties  es.properties  it.properties        si.properties
ch.properties  fi.properties  keyboard.properties  sv.properties
cs.properties  fr.properties  no.properties        tr.properties

Output of 'ls /dev/hidg*'
ls: cannot access /dev/hidg*: No such file or directory

jairwen, 2018-04-11:
In my case, conncted from WIFI, 
`
Linux MAME82-P4WNP1 4.9.78+ #1084 Thu Jan 25 17:40:10 GMT 2018 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Apr 11 10:34:40 2018 from 172.24.0.67
suThere is no screen to be detached.
pi@MAME82-P4WNP1:~
`
manual launch it.
`sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py
Traceback (most recent call last):
  File "/home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py", line 27, in <module>
    from pydispatch import dispatcher
ImportError: No module named pydispatch` 

Seems Installation is not complete. I will reinstall it again.

mame82, 2018-04-11:
You could use the pre-build image from the release

jackinchy, 2018-09-11:
Turns out for me also that a faulty USB cable was the issue. I would recommend swapping it out if you have any problems of the same sort.