RoganDawes/P4wnP1 #27: Troubles when installing via instructions given in INSTALL.md

Swiftb0y, 2017-07-31:
Ive tried installing P4wnP1 headless with the instructions given in the Install guide.
I get as far as SSHing into the PI, but the connection just breaks after a few seconds and ssh as well as ping report that the Pi is unreachable.

Any help would be greatly appreciated.

battletux, 2017-07-31:
Have you tried installing this via a Raspberry Pi 3?

I was able to install it via  RPi3 with no issues. Write Raspbian Jessie Lite to the SD card, pop it in the RPI3 and then clone the git repo and run the installer.

Once completed shutdown the Pi3 and put the SD card in the RPi0 and you are golden.

Swiftb0y, 2017-07-31:
No. But you reminded me that Ive got a RPi2 with a broken HDMI port.
I connected that one via ethernet, ssh'd into it and got it to work. thanks

closed 2017-07-31 by Swiftb0y

mame82, 2017-07-31:
Wow, issues get solved before I am even able to read them.

Anyway, the most likely reason for the SSH disconnects is that the usb network interface on the host you're using has a running DHCP client attached. This would overwrite the IP you configured for the interface after some seconds (if no DHCP lease is given by the Pi, the IP is wiped out)