RoganDawes/P4wnP1 #31: Hidden SSID possible?

drifter213, 2017-08-01:
Hey,
I just wanted to ask, if it is possible to hide the SSID from the AP?

mame82, 2017-08-01:
Sorry, no time to test this at the moment, but it should be simple.

Read this
https://www.quora.com/Is-there-any-way-to-hide-ssid-broadcast-on-AP-mode-wi-fi-hotspot-in-Linux

... and do it here
https://www.quora.com/Is-there-any-way-to-hide-ssid-broadcast-on-AP-mode-wi-fi-hotspot-in-Linux

Feedback appreciated! 

If it works I'll add in a setup option

drifter213, 2017-08-01:
Thank you for your fast response!

I will try it in a few hours... currently not at home.

I will keep you updated

mame82, 2017-08-02:
> Even if I set it back to 0, I have to redo the image, so the AP is working again.

Weird. Would have been nice to have a hidden AP. Maybe I come back to this, when I have time to.

Going to close the issue for now.

Thanks for testing, anyway

drifter213, 2017-08-02:
Hey, it's me again.

just tried it again, now it worked instantly -> ignore_broadcast_ssid=1

I must have accidently deleted something in the init_wifi.sh file.

mame82, 2017-08-02:
Nice, so this should be put on the ToDo list, for a setup option. I have to delay implementing this, cause I'm working on fileupload functionality for the backdoor and stumbled across various minor bugs which need to be fixed first

whackyhack, 2017-08-03:
For anybody with outdated client, ignore_broadcast_ssid=2 might be easier.

mame82, 2017-09-04:
`WIFI_ACCESSPOINT_HIDE_SSID` option to hide SSID implemented in setup.cfg:

https://github.com/mame82/P4wnP1/blob/master/setup.cfg#L44

Note: Method 2 is used (clear SSID with ASCII 0), instead of option 1 (use SSID of length 0).

closed 2017-09-05 by mame82