RoganDawes/P4wnP1 #33: covert feature isn't being covert??

mrhut10, 2017-08-02:
Plugged into a new windows 10 machine, 
payload is the covert HID and although sometimes works i.e. get a connection.

The covert feture isn't working, have a window on screen displayed also with the errors 

please see pastebin for the errors / output of the window
[www.pastebin.com/WfVAAzNH](url)

anyone else seen this?

mrhut10, 2017-08-02:
p.s. on my win10 laptop works perfectely, so must be something configured differently on this new win 10 installation.

( maybe its because developer options not turned on? although i have no clue )

mame82, 2017-08-02:
That's weird. The two powershell CmdLets `Get-Process` and `New-Object` aren't present when the powershell sessions starts. Even more weird: the second part of the payload use `New-Object` again, without issues.

So rough guess: the powershell runspace hasn't finished loading the needed assemblies for `Microsoft.PowerShell.Utility` and `Microsoft.PowerShell.Management`.

As I'm not able to reproduce this could you please test increasing the delay between powershell console start and input start to 5 seconds with:

`FireStage1 1 5000`

mrhut10, 2017-08-02:
I'll try increment it now, however first test of 5 seconds didn't appear to make a difference.

its almost a complete fresh install of windows 10 also, so this is likely the first power-shell script to be run on it.

mrhut10, 2017-08-02:
i jumped straight to a 10 and 20 second and didn't make a difference either?
` FireStage1 1 10000`
` FireStage1 1 20000`

and neither worked covertly as intended.

basic specs, brand new HP computer with an i7 processor so wouldn't have guessed its performance was stopping it, but obviously I've no idea either.

Any other details machine or operating system that would be helpful?

also i just opened a new powerhsell window, and typed
`Help`
which reported a similar error of a module not being found.

but then when i typed it a second time in the same window it worked fine the second time???

mrhut10, 2017-08-02:
i just manually typed the first command manually from the payload in a new powershell,

failed as expected, however when i then typed it again a second time as long as i did in the same window it worked fine??? really strange, seems that on my computer i need to run some commands twice for them to work.

no one else obviously has had a similar problem??

mrhut10, 2017-08-02:
i now suspect may be the McAfee live safe silly antivirus that came with it, i'm currently uninstalling to see if it fix's, will report result either way.

mame82, 2017-08-02:
It seems like some modules are missing on start, what's the output of 
`Get-Module -ListAvailable` in a new powershell session?

mrhut10, 2017-08-02:
ok, resolved.!!!!!!

appears that antivirus had stuffed up somehow loading modules.

closed 2017-08-02 by mrhut10

mame82, 2017-08-02:
Hmm, this is a interesting use case, because McAfee seems to hinder execution of basic powershell commands. This payload isn't meant to be stopped by AV. Unfortunately I have no test environment available to dive into this