RoganDawes/P4wnP1 #36: john-1-8-0-jumbo_raspbian_jessie_precompiled folder empty

vdub804, 2017-08-05:
thanks for such great work really enjoying tinkering with my pi0w which has been collecting dust until now. I manually downloaded the  john-1-8-0-jumbo_raspbian_jessie_precompiled folder and extracted the tar, am I supposed to place the complete john folder inside of john-1-8-0-jumbo_raspbian_jessie_precompiled or just the contents from the john folder? thanks!

mame82, 2017-08-05:
If you're using P4wnP1, the extracted content is expected in `P4wnP1/john` folder. But extracting should already have been done by the installer (as long as you cloned P4wnP1 recursivly). The onöy payload using john is LockPicker, right noe

vdub804, 2017-08-05:
ah ok that's where I screwed up I didn't clone recursively my apologies

closed 2017-08-05 by vdub804