RoganDawes/P4wnP1 #38: Adding own payloads/using the established connection/metasploit integration.

Swiftb0y, 2017-08-06:
Just a question... xD

mame82, 2017-08-07:
Could you get more precise

Swiftb0y, 2017-08-07:
While a regular dos shell is pretty useful, it would be nice to use a meterpreter shell for post exploitation. 
I'm not very familiar with this stuff but this is how I would do this:
-add metasploit as a submodule.
-enable HID and RNDIS (and CDC ECM if linux support gets added later.)
-start the console, generate the webdelivery psh stager and serve it (all in a screen session)
-detach screen
-connect to screen onLogin.
-boom everything there. with every post exploitation module as usual.

The only catch would be the IEX thing.

My other suggestion is there should be the option to start as admin by using the start menu to open powershell and pressing ctrl+shift+return to open UAC and spamming [left]+return, which opens cmd/powershell (wasnt able to confirm it with powershell yet) as admin.

mame82, 2017-08-07:
That's perfectly doable with P4wnP1, in fact it has been done with an older version at Black Hat Session XV:

https://www.madison-gurkha.com/hands-onhacking-raspberrypi-en

Swiftb0y, 2017-08-08:
ok. Will you integrate that or should I do it myself and PR it? And how would one develop that?  Because once the P4wnP1 installation is done, there is not really a way to connect it to the internet (without extra hardware like a usb-ethernet adapter). Can I configure my existing installation so it can connect to my home wifi instead of opening one? (this would also be a useful feature to extend the range. So the Pi could connect to a repeater/router which extends the range as well as offer the possibility of using the it as a working ethernet adapter (probably to much work, though)).

mame82, 2017-08-08:
Regarding a payload connecting as WiFi client to a given network, I'm fine with a PR.

Regarding a payload which includes metasploit components, I won't accept a PR.

Not that I won't like to, but I try to avoid dependencies to external project, because of the different copyleft licenses they use.

As I'm not a lawyer, I'm in fact going to remove the Responder and JtR submodules in the next major, to have the freedom of choosing the best license for P4wnP1. This again implies adding in a payload, which allows the user to install missing dependencies on its own and thus connect to the internet.

Swiftb0y, 2017-08-08:
thanks. I guess I will do the metasploit thing in my own fork.
The problem with the wifi client is that im not that familiar with the configuration and where to change what, so it would take some time until I am familiar enough with the code...

closed 2017-08-08 by Swiftb0y