RoganDawes/P4wnP1 #39: Unable to set German Keyboard Layout

felbinger, 2017-08-07:
Hello,
I am unable to set the German keyboard layout. I tried it with `SetKeyboardLanguage de`. If I type `SetKeyboardLanguage`, I see that it are bracketed arround it. 
`
Choose language by number or name:
================================

0:br    1:dk    [2:de]  3:ch    4:ru    5:be    6:it    7:cs    
8:hr    9:es    10:fr   11:tr   12:ca  13:si   14:sv 15:pt   
16:no 17:gb 18:us   19:fi
`
I guess that this feature will be implemented soon, right? Can I help? Tell me what is to do that the feature get implemented and I try to help. I am not the best programmer but I will give my best.

Best regards
Nico Felbinger

mame82, 2017-08-07:
It is implemented and tested. In fact I'm using de keyboard layout myself. Is duckencoder.py installed correctly  (cloned recursively) ?

felbinger, 2017-08-07:
Okay, I installed it again on a fresh Raspbian installation and its working now.

closed 2017-08-07 by mame82