RoganDawes/P4wnP1 #41: There is no screen to be detached -HID Backdoor

jgbuddy, 2017-08-07:
this is a great project thanks for all the hard work and dedication mame82.

I'm also getting the same "there is no screen to be detached" error, even after trying multiple cables. If I run ~\P4wnPi1-master\hidtools\backdoor\P4wnP1.py, i get :

File "P4wnP1.py", line 13, in <module>
    from DuckEncoder import DuckEncoder 
ImportError: No module named DuckEncoder

This is probably some stupid mistake on my part as i am new to this stuff, but any help would be much appreciated! Thanks!

mame82, 2017-08-07:
Sounds like you didn't cloned recursively and are missing duckencoder.py

jgbuddy, 2017-08-07:
ok, googled it and everything makes sense now. thanks

mame82, 2017-08-07:
No need to use google, everything is here https://github.com/mame82/P4wnP1/blob/master/INSTALL.md

;-)

closed 2017-08-07 by mame82