RoganDawes/P4wnP1 #43: Restart P4WNP1 Server

Luuk2016, 2017-08-08:
If i close the server using CTRL+C is there a way to reopen the server. Everytime i do this i need to restart the Pi Zero W

mame82, 2017-08-08:
Instead of closing the server (which is better done with the `exit` command), issue `ctrl+a` followed by `d` to detach the screen. Run `sudo -d -r` to re attach it to a different terminal (or the same terminal in you case)

closed 2017-08-08 by Luuk2016

jcstill, 2017-09-13:
@mame82, not sure if you could help, when I use the `ctrl+a` and then `d` I get:
`*** Unknown syntax: EOF`

mame82, 2017-09-13:
@jcstill
I'm not able to reproduce this behavior. Maybe a terminal issue of the SSH client in use.

What are you using to SSH in?

jcstill, 2017-09-14:
The default terminal on kali.

mame82, 2017-09-14:
Seems to be related to this (cmd.py (

https://bugs.python.org/issue13500

But I'm still wondering why this doesn't happen to me.

Maybe I should have a look if the escape sequence of `screen` could be changed, don't want to do changes to cmd.py !