RoganDawes/P4wnP1 #44: Can't get the P4wnP1 to work with raspberry PI zero W

jonas18z, 2017-08-09:
Raspberri pi zero W 
1. Installed raspberry jessie lite with etcher 
2. add SSH (empty )  file and wpa_supplicant.conf  ( to my home wifi) in boot folder.
3. Connected via SSH and runned: 
sudo apt-get -y install git
cd /home/pi
git clone --recursive https://github.com/mame82/P4wnP1
cd P4wnP1
./install.sh

4. Rebooted and changed port to USB to closet on to HDMI-port and put it in a Windows10 computer USB. 

5.I connect from another computer to WIFI P4wnP1 with password MaMe-P4wnP1
6. Edit the setup.cfg  added # to network only and removed from backdoor. 
7  Pulled power and again.
8 connect again to wifi and  172.24.0.1.  and get the message "
There is no screen to be detached." 
9. Just the normal linux shell and nothing on man Windows computer happedns. Tryed every payloads but noone works. 

Reinstalled everything 3 times and did the same step. I don't have a monitor to plug in the HDMI and I don't have any USB-to ethernet. 

What have I done wrong?

mame82, 2017-08-09:
According to your descridescription, it seems you haven't done anything wrong according th install process.

Could you please give some details on what drivers have been installed by Windows (what is detected in device manager). Additionally the output if `uname -a` ftom P4wnP1 SSH shell.
Please assure that your USB capable is working, too.

jonas18z, 2017-08-09:
Daym! 

I checked my cable and I had a cable that is "microusb charger cable" and not the data pins a connected!  The pi got power but no data! So stupid with new cable it's working!

closed 2017-08-09 by mame82