RoganDawes/P4wnP1 #45: 3rd party tools as plugins/addons

Swiftb0y, 2017-08-12:
Ive already asked about metasploit integration which could be difficult with copyleft, so my Idea was that tools like metasploit, responder, jtr, etc. aren't directly integrated in the form of submodules, but rather installed elsewhere and the installation just contains scripts to interface with these applications.

I don't know if that goes ok with licensing and so on, after all, Just an Idea... :)

mame82, 2017-08-13:
Ongoing discussion from https://github.com/mame82/P4wnP1/issues/38

mame82, 2017-08-13:
Yes, I was thinking about this. As P4wnP1 is based on Raspbian, package managers like `pip` and `apt` are available and do exactly that.

In fact, an early version of P4wnP1 used an installer script which adds in external tools via `git`. 

From my opinion this would be the way to go. But when it comes to JtR Jumbo for example, this means a script would be needed downloading the source, installing the compiler chain and dependencies, fix the source to make it compile on Raspbian and then start compiling (takes about an hour). If something goes wrong, the user is on its own with this. 

So I still haven't found the best solution.

Swiftb0y, 2017-08-13:
Im working on a prototype with metasploit where you point to the msfdir in the setup.cfg (similar to the SE toolkit) and a script that has a bunch of functions that spawn a listener, attach to the console, etc. 
They should just act as an Addon (so P4wnP1 itself doesnt depend on it) and are not included in the installation. Just because the automatic installation with rvm and postgres setup can be diffucult to automate as well.
Would you accept a PR once I've got a working prototype?

mame82, 2017-08-13:
I'd suggest creating a dedicated payload, pointing to the msfdir, from there you could use relative path to P4wnP1 working dir with `$wdir` or use an absolute path. Install instructions could be added in as comment or a dedicated script could be used. You could create a whole subdir in `payloads` folder, if needed.
Proper MSF installation could be checked by the payload and `led_blink` command could be used to indicate an error, if needed.
PR  requests for payloads are the prefered option.

mame82, 2017-08-13:
Your work on the WiFi client option and USB mass storage size are considerable features for a PR in P4wnP1 base functionality (not payload). 

For USB mass storage, I had some things in mind myself, like:
- CD-Rom emulation with given USO image
- mounting flash drive image to P4wnP1 FS when not used (for example hashes grabbed by LockPicker payload could be stored to the flashdrive image and another payload could be used to remount it as USB flashdrive, to give access to the stored data)

Swiftb0y, 2017-08-13:
Ill see what I can do.

My problem with the Wificlient is that it only works somewhat. Ive added a function to init_wifi.sh which works when run after the accesspoint was set up, but not before, and I dont know which steps I have to add. I can PR it but you would have to debug it since I lack the necessary networking knowledge.
I also cant get the forwarding to work any idea?

Ive also added a bit of code to the installation script which generates an Image that is 4/5 of the available space on the sd card. Problem is that this takes pretty long and dd is still version 8.23 and doesnt know about the status=progress flag (added in 8.24). So in my case with a 32gb SD card, it takes pretty long and without output it might irritate the user...

Swiftb0y, 2017-08-15:
I think if got a almost working prototype but its difficult to debug without logs, do you know where they are stored?

Swiftb0y, 2017-08-18:
For some reason, starting screen in detached mode doesnt really work. The same method is used in the responder payload:
148: `screen -dmS responder bash -c "cd $wdir/Responder/; python Responder.py -I $active_interface -d -r -w -P"`
When I try to start msfconsole with `screen -dmS msfconsole -bash --login -c "cd $MSFPATH;./msfconsole -L"` nothing shows up after boot.
(-L is there because screen terminates otherwise)

mame82, 2017-08-18:
Remove the `-` in front of bash.

Swiftb0y, 2017-08-18:
xD that just a type I made right now, thats not in the code.

mame82, 2017-08-23:
You should omit the `--login` parameter, too. According to bash manpage this relaces the current shell, but screen opens an own tty

Swiftb0y, 2017-08-26:
I cant try it out right now, but what do you think of my PR?

mame82, 2017-08-27:
Commented on PR

closed 2017-08-27 by mame82