RoganDawes/P4wnP1 #46: WIFI Access Point does not show up

tomisboy, 2017-08-12:
I have a Raspberry Pi Zero w 
I installed everything like in your install.md 
With the command ifconfig i can see that my pi has wlan0 and its active, 
however i cant find the wifi . Itest it with an Andoid Phone and my Win7 Laptop

gloglas, 2017-08-12:
Yeah i got that issue too. And i think that my Pi0W is stucked in interactive mode even if i turn off hdmi and enable dwc2 and g_ether. I spend 4 hours of working on it..

mame82, 2017-08-12:
Interactive mode should only boot if an USB OTG wire is attached. Use a normal USB cable to ise P4wnP1 in device mode.

tomisboy, 2017-08-12:
I used a normal micro USB Kabel Like you would load your mobile phone

gloglas, 2017-08-12:
After couple hours of wondering. I setup it all again, tested 4 cabels, 2 win7 computers. Try disabling/enabling dwc2 and g_ether. But not even hotspot shows up.
Do you have pre-builded P4wnP1 img? It will be awesome.

mame82, 2017-08-13:
@gloglas
https://github.com/mame82/P4wnP1/blob/master/FAQ.md#the-setup-isnt-easy-will-there-be-a-prebuilt-image

Please provide more details, which payload have you choosen, what's the output on boot when P4wnP1 is attached in device mode (HDMI).

gloglas, 2017-08-13:
Ou how I can miss this ... The hostpad, screen and dnsmasq is not preinstalled in jessie lite. But how i can get ethernet work again?

mame82, 2017-08-13:
@gloglas

That's all done by the P4wnP1 installer. Best advice i could give, is to restart with a clean raspbian image and follow the instructions in INSTALL.md

gloglas, 2017-08-13:
No it is not. Ok fifth round of cleaning.

mame82, 2017-08-13:
Please watch the output of install.sh, if you're able to spot possible errors I could maybe assist. Be sure the internet connection is working before running `install.sh`.

Please provide the output of apt-get, when it comes to package installation.

luke-spademan, 2017-08-13:
try running install.sh as root:
`sudo ./install.sh`

That did it for me

gloglas, 2017-08-13:
Yeah I always do sudo when it comes to installing something. I installed hostapd... before and now it works! But only network_only. When i enable hid_backdoor it brokes. Those are photos from last try.
Btw I am on really bad internet.
[image]
[image]
[image]
[image]

After enabling hid_backdoor
[image]
[image]

gloglas, 2017-08-13:
Few more..
[image]
[image]

MrGrimod, 2017-08-13:
For WPA2 your password needs to hold a certain amount of characters. 
Are you using the default password?

mame82, 2017-08-14:
@gloglas 

Thx for your screenshots, they help to nail down the problem:

1. The first screenshot shows the successful install of pycrypto. Although a screenshot for the apt-get running upfront is missing, I assume this was working without issues, otherwise `pip` wouldn't be available at this point. So there shouldn't be a need for manual installation of hostapd and dnsmasq, as both packages are installed along with pip.
2. There's no need to run the installer with sudo, as every privileged command is prepended with sudo. Doing so would mess the permissions, as files meant to have writable privs for user `pi` would end up owned by root.
3. Your last install screenshot shows, that rpi-update fails with an SSL error. This again indicates that your internet connection gets lost during kernel update.
4. The missing kernel update is the reason for the failing payloads (kernel error output after reboot) because the USB gadget drivers on kernel 4.9.35+ aren't working.

Conclusion:

Assure your internet connection is up during whole install process. If you followed my steps for bringing up internet headless, the most likely reason for losing internet access: Kali tries to reassign an IP address to the USB network adapter via DHCP. When you manually configure an IP to the adapter, be sure it hasn't got a DHCP client running. The quick and dirty fix on kali would be `service network-manager stop` which prevents automatic reconfiguration of any kali network interface. More clean would be to configure 172.16.0.2 as static IP for usb0 on kali. I need to update install instructions on this.

To solve your problem, use the network-only payload. Attach P4wnP1 to a Windows host. Enable ICS to the RNDIS adapter. This would reassign a IP to the RNDIS adapter, wbich needs to be reset to 172.16.0.2 manually (while ICS still enabled). SSH into Pi `pi@172.16.0.1`. Add the proper degault gw to Pi with `sudo route add -net default gw 172.16.0.2` 
Your pi should have internet access now (assure with `ping 8.8.8.8` for example). If DNS isn't working, add a nameserver to `/etc/resolv.conf`. After internet access if given, rerun `sudo rpi-update`, which should fix the kernel error.

The easier way would be a reinstall, but make sure your internet stays up. If you loose connection again, fix it and rerun install.sh BEFORE REBOOT (no sudo).

Hope this helps

gloglas, 2017-08-14:
So i don´t know what happend over night i just loaded my backup with downloaded P4wnP1 before installation. I did not use **sudo** and i think that was the biggest problem (maybe add warning to INSTALL.md). The installation was faster than before so I do not know if it is by it or internet. Works how it suposed to. 
And one more question how to change vendor and product id "right way"?
Btw your project is amazing! Works even with CS keyboard language (i didn´t expect that). And thanks for all the support!

mame82, 2017-08-17:
VID and PID should be changed in the respective payload, like here

https://github.com/mame82/P4wnP1/blob/master/payloads/hid_keyboard.txt#L13

The LockPicker payload shows how to use a random PID:

https://github.com/mame82/P4wnP1/blob/master/payloads/Win10_LockPicker.txt#L63

tomisboy, 2017-08-17:
can be closed . 
I bought a USB to LAN adapter 
Installed everything new
did apt-get update before
now its working fine

mame82, 2017-08-17:
Nice!

closed 2017-08-17 by mame82

mr0Ot, 2017-12-12:
Hi a have a question.
when i will start the: "PAYLOAD-hid _backdoor.txt "

-bash: /tmp/blink_count: Permission denied 
There is no screen to be detached.﻿

Please can you tell me waht i make wrong?
i understand not what i must do.
it will make premission, ok, what must i do?

THX for the help