RoganDawes/P4wnP1 #48: Internet cable isnt working over micro usb data port.

Luuk2016, 2017-08-15:
Currently i have P4wnP1 installed and i like it sofar.
Now i want to do some more things using my Pi Zero W, like updating and installing some more packages.

I wanted to connect my Pi Zero W to the internet using a ethernet to usb to micro usb cable.
But it isnt working. My Pi doesnt show up in the network.

My only fix now is to first install all other packages and then install P4wnP1.

Hopefully you can help me solving this problem

mame82, 2017-08-15:
Let me suggest to use the `network-only.txt` payload and use Windows Internet Connection Sharing to bring your Pi online.

More detailed instructions are given in the following comment (part avout ICS)
https://github.com/mame82/P4wnP1/issues/46#issuecomment-322127650

Luuk2016, 2017-08-15:
Could i also just change the payload temporary to network only? THen update and then change back.

mame82, 2017-08-15:
Yes, but you have to follow the given instruction to obtain internet access, because 1) P4wnP1 doesn't know the needed gateway IP (it runs a DHCP server, not a client) so a route has to be configured 2) Your computer needs to be setup to route P4wnP1 traffic out to the internet (easiest way on Windows is to use ICS)

mame82, 2017-08-16:
I've uploaded a little tutorial, to give you some guidance on this. Hope it helps

https://youtu.be/QEWaIoal5qU

Luuk2016, 2017-08-16:
Ok thanks. So everytime i have to update/download something from the internet i have to change the payload to network_only. The thing you explained in the video have to been done 1 time only on the attackers pc, right?

Luuk2016, 2017-08-16:
I am mostly using the hid_backdoor payload

mame82, 2017-08-16:
> Ok thanks. So everytime i have to update/download something from the internet i have to change the payload to network_only. The thing you explained in the video have to been done 1 time only on the attackers pc, right?

Yes, as long as you don't change PID and VID. But the gateway route has to be added every time. Maybe this should be added to the network only script.

closed 2017-08-25 by mame82