RoganDawes/P4wnP1 #49: SetKeyboardLanguage be

Skrychi, 2017-08-16:
Setting this keyboard layout crashes the interface and shoves you straight to the terminal. I haven't checked other layouts, but so far i've tested fr, de, us, gb, and es. So its possible that other layouts are affected?
I've tried reformatting and starting from scratch multiple times with the same result.
Please fix :')

mame82, 2017-08-16:
I'll investigate this, but I'm afraid I'm not able to work on this the next two weeks.
So please be patient (filed as bug already).

julienlemonde, 2017-08-19:
To add to the bug, same behavior for ca.

mame82, 2017-08-25:
@julienlemonde  @Skrychi 

ca, be, ru and no had errors, which are hopefully fixed in latest release.

Could you confirm

Skrychi, 2017-08-25:
So far, sending ducky scripts work fine.
Here's what I get when I trigger FireStage1
As you can see bottom right i've tested it on all 3 be layouts.
http://imgur.com/a/yoVMc
[image]

mame82, 2017-08-25:
`[` and `]` are excchanged...so still an issue in be.properties

julienlemonde, 2017-08-28:
Ca is working now, Thanks!

mame82, 2017-09-01:
@Skrychi 

BE layout should be working now (tested with `FireStage1`)

Skrychi, 2017-09-01:
Fixed. As for the others, Someone will have to confirm.

closed 2017-09-01 by Skrychi

MaximusBaton, 2019-03-13:
sorry for reopening this issue, however I'm facing the same problem.
the victim's default windows language is 'ru'. so the FireStage just types ru symbols in powershell window.
changing the language to us with SetKeyboardLanguage perform an app crash like the topic started had described.
u have any clue for this?