RoganDawes/P4wnP1 #50: Control via bluetooth

Yamakaky, 2017-08-16:
https://hacks.mozilla.org/2017/02/headless-raspberry-pi-configuration-over-bluetooth/

It would allow the Pi to connect to a Wifi network instead of hosting one. In addition of standard internet access for update, it would allow the Pi to act as a USB ethernet with transparent monitoring.

mame82, 2017-08-16:
I definitely going to implement BT support (at least for backdoor payload).
I thought about it already, but had something like PAN networking in mind, to allow SSH access (Should be supported by ConnectBot from most android devices, for example).

Anyway, it will take some weeks to implement it. Priority features right now are file upload/download for backdoor payload, followed by socket connections tunneled through HID backdoor (another reason not to switch from SSH to pure serial shell, as SSH allows TCP tunneling).

Thanks for the request, let's see how long it takes (PR welcome ;-))

mame82, 2017-09-20:
Bluetooth support added in latest commit (NAP)

closed 2017-09-20 by mame82

Yamakaky, 2017-09-21:
Yeay, will try!

Commander-Crash, 2019-06-09:
I like to implement the PAN bluetooth network how you did with P4wnP1 to another rpi3 running raspbian.

commit 812accf7 (2017-09-20): Merge branch 'bluetooth_dev'

Merged in bluetooth support, fix #50