RoganDawes/P4wnP1 #53: No screen to be detached

asing012, 2017-08-21:
Payload - hidbackdoor
Target OS - Windows 7

I read the other two tickets about this same issue and I tried everything but i am still getting no screen to be detached error. Keyboard driver seems to be installed in the target OS and I am able to run the P4wnP1.py using python command. But when I type FireStage1 it's not doing anything.

mame82, 2017-08-21:
Please chech the content of the `duckencoder` folder. Are there any files ?

asing012, 2017-08-21:
It has the duckcoder.py README.md and resources folder.

mame82, 2017-08-21:
Okay, no problem with encoder.
Why do you have to start `P4wnP1.py` manually, is there an error on boot (HDMI out).

Please also check the output of `ls -la /dev/hidg*`.
Make sure `hid_backdoor.txt` payload is selected, first.

asing012, 2017-08-21:
I found the issue, It was the cable. It is working now..
Thank You

mame82, 2017-08-21:
Nice to hear

closed 2017-08-21 by mame82

TeCoEd, 2017-09-24:
My issue was the cable too, tried a USB Stem and did not work, went back to old data cable and worked fine

mame82, 2017-09-24:
Added a USB Stem right now and it works like charm for me

https://twitter.com/mame82/status/911879277230075904

TeCoEd, 2017-09-24:
Odd as this is the exact same one I used, must be my poor soldering skills!
Program has 'screen detached' error run on USB stem, but use cable and it works fine!