RoganDawes/P4wnP1 #54: automatically run ducky script on plug in

andrew1999, 2017-08-21:
How do I set it up so it runs a ducky script automatically upon plug in?
i tried editing the hid_keyboard.txt file in /P4wnP1/payloads but every "$" sign in the code causes issues.

mame82, 2017-08-21:
`duckhid` is a bash funtion accepting output from STDIN (f.e. from a pipe `|`). The payload itself is a bash script. If you use it like it's done in `hid_keyboard.txt` using the `cat` command with a multiline string, you have to escape `$` signs with `\$` (otherwise bash would try to resolve a bash variable).
A cleaner way would be to store your script on the `DuckyScripts` folder and use `cat $wdir/DuckyScripts/yourscript.duck | duckhid`.

The nice thing about the second approach is that you can run multiple duckyscripts, while keyping the payload clean.

mame82, 2017-08-23:
Is your problem solved?

andrew1999, 2017-08-23:
Yes, thank you!

closed 2017-08-23 by mame82