RoganDawes/P4wnP1 #58: Will not mount HD Keybaord, no Screen to be detached, P4wnP1 server wont start.

mkbcaptain, 2017-08-23:
Target is Win 7 and 10.

There is no screen to be detatched error and Device is not showing up as HD Keyboard, I can ssh headless and via wifi. Please help.

Output of 'sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py'
Traceback (most recent call last):
File "/home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py", line 855, in
HIDin_file = open(dev_file_in_path, "rb")
IOError: [Errno 2] No such file or directory: '/dev/hidg1'
Cleaning Up...
Traceback (most recent call last):
File "/home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py", line 891, in
ll.stop() # send stop event to read and write loop of link layer
NameError: name 'll' is not defined

John folder is full of stuff

Output of 'ls -R /home/pi/P4wnP1/duckencoder/'
/home/pi/P4wnP1/duckencoder/:
duckencoder.py README.md resources

/home/pi/P4wnP1/duckencoder/resources:
be.properties de.properties gb.properties pt.properties us.properties
br.properties dk.properties hr.properties ru.properties
ca.properties es.properties it.properties si.properties
ch.properties fi.properties keyboard.properties sv.properties
cs.properties fr.properties no.properties tr.properties

Output of 'ls /dev/hidg*'
ls: cannot access /dev/hidg*: No such file or directory

mame82, 2017-08-23:
Seems am error with HID gadget kernel module.
What's the output of `uname -a`

mkbcaptain, 2017-08-23:
Linux MAME82-P4WNP1 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux

mame82, 2017-08-23:
Something went wrong during installation, as the latest kernel update isn't installed. Seems you lost internet connection while running `install.sh`.

If you connect the Pi to internet, you could install the update with `sudo rpi-update`.

But I don't

closed 2017-08-23 by mame82

mkbcaptain, 2017-08-23:
Don't what?

mame82, 2017-08-23:
Refresh the page ;-)

mkbcaptain, 2017-08-23:
HA!  Gotcha, Okay, I have done a clean install twice now, maybe third time is the charm.
Thanks, Will keep you posted.

mame82, 2017-08-23:
With which method did you connect to Internet?

mkbcaptain, 2017-08-23:
Headless via ssh, shared connecting with laptops LAN.

mame82, 2017-08-23:
If you manually hand out an IP to the network adapter which represents the P4wnP1 network interface, be sure there's no DHCP client running, as this would wipe the manual IP configuration after some time.

Watch the output of `install.sh` closely. In the end `rpi-update` is ran, if there is an error, it is likely you lost internet connection.

mkbcaptain, 2017-08-24:
Okay, so I just flased an SD card with Jessie lite, updated and upgraded, ran 
sudo apt-get -y install git
cd /home/pi
git clone --recursive https://github.com/mame82/P4wnP1
cd P4wnP1
./install.sh

and now the output of  uname -a is 

Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

Only "errors" I saw.
[image]

No sure what to do at this point.

mkbcaptain, 2017-08-25:
No ideas on this?

mkbcaptain, 2017-08-25:
So after manually running rpi-update the out put of  `uname -a` is

Linux MAME82-P4WNP1 4.9.44+ #1029 Tue Aug 22 16:41:39 BST 2017 armv6l GNU/Linux

So it looks like it worked, the problem I was having, I was using a Jessie lite version that didn't have the rip-update installed, so I had to run `sudo apt-get install rpi-update`. 

Testing further now

mkbcaptain, 2017-08-25:
Okay, Still not working, was using Jessie lite 2017-04-10, trying again now with Jessie lite 2017-07-05.

mkbcaptain, 2017-08-25:
Okay, So Jessie Lite 2017-07-05 worked perfectly fine, however on initial driver install it wouldn't work on target computer if I plugged the pi in via the soldered USB to the Zero W. But if I plugged the cable into the Zero W and the target computer it worked fine, unplug then plug back in via the soldered USB connection it works.  Thoughts?

mame82, 2017-08-25:
No thaughts on the last one, but I know that the default `g_ether` module of raspbian gets detected as serial COM port on Windows 10, instead of RNDIS. With P4wnP1 installed it gets detected as RNDIS again.

mkbcaptain, 2017-08-25:
HID-keyboard works, win 10 lockpicker works, but the HID-Backdoor still says no screen to be detached, when `sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py` it works but says no client connected. and `FireStage1` outputs "Starting to type out stage1 to target...."
and then never does anything.
`FireStage1 1 5000 nohide` outputs "Starting to type out stage1 to target...."