RoganDawes/P4wnP1 #59: Shell disconnects for some/this reason

Anonyvius, 2017-08-23:
Everytime i click for example on an youtube video i get this error and the client disconnects:

C:\Users\User\Desktop>Client received terminates!
P4wnP1 shell (client connected) >
Target disconnected
P4wnP1 shell (client not connected) >
Client disconnected, stop interacting
P4wnP1 shell (client not connected) >
Client didn't respond for 1 seconds.
P4wnP1 shell (client not connected) >

i dondt know why. It works i can use the shell but if the victim click for example on an youtube video i get disconnectet.

Sorry für mein schlechtes englisch.

mame82, 2017-08-25:
Could you give more details on what you mean with 

>  i can use the shell but if the victim click for example on an youtube video i get disconnectet

The shell disconnects, if the underlying communication layer doesn't send data for more than a second.
This should only happen if the agent started on the client (brought up by `FireStage1`) terminates or stalls for some reason. But I doubt that YouTube would block execution of a PowerShell process

mame82, 2017-08-27:
Any progress on this?

Anonyvius, 2017-08-27:
Not realy. I mean i have the problem that i the victim open an programm or do something on the PC the shell disconnect.

Sorry für mein schlechtes englisch. was ich sagen wollte. wenn das opfer am pc zB. ein Programm öffnet oder auf zB. YouTube eine Aktion ausführt werde ich einfach herausgeworfen.

Anonyvius, 2017-09-05:
Okay. Update: Now i habve the problem that the shell disconnects every time after like 10 seconds.

PoSHMagiC0de, 2017-09-09:
@Anonyvius 
Try using FireStage1 with the nohide argument and see what the client on the victim is doing or saying when it happens.  You maybe able to get a useful error message on the victim end.

mame82, 2017-09-26:
Any progress @Anonyvius ?

mame82, 2017-10-10:
Ussue could be reopened if problem remains

closed 2017-10-10 by mame82