RoganDawes/P4wnP1 #60: Win10_Lockpicker takes forever

Anonyvius, 2017-08-23:
I dont know why. But the Win10 Lockpicker does not work i plugged the raspberry with the payload in my PC. Now i am waitung since 1 hour and nothing happens. The LED flashes two times. That meas it is waiting for the hash but nothing is happining.

mame82, 2017-08-25:
That should be the normal behavior. Anyway, for me it worked on every testcase.
Unfortunately I'm not allowed to give any details on this (and thus I'm not able to tear down the problem).

I recommend reading the comments in the `Win10_LockPicker` payload file, perhaps they could point you in the right direction.

amohacker, 2017-08-27:
My laptop just blocks the connection with the message isa policy has been changed and on my desktop a window pops up in my browser saying log in to proxy authtome:3128

mame82, 2017-08-27:
@amohacker 
This is because HTTP is redirected to a non existing proxy server by Responder, which asks for authentication. Grabing hashes through manual HTTP access isn't in scope of the payload.

It's meant to be used on locked machines with a user logged in.

As stated, I'm not able to give more details, right now.

closed 2017-08-27 by mame82