RoganDawes/P4wnP1 #62: Not sure what I did wrong

jcstill, 2017-08-24:
@mame82 such a sick project. I am running into a problem with the HID backdoor. 

#1
when the pi boots, i get the "There is no screen to be detached" message. I run "sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py" and it starts up just fine. The output of "uname -a" is: 

Linux MAME82-P4WNP1 4.9.44+ #1029 Tue Aug 22 16:41:39 BST 2017 armv6l GNU/Linux

Thanks for the Help
-J

Swiftb0y, 2017-08-24:
I dont know what your issue is either, but did you make sure you were using raspbian jessie and cloned with --recursive?

mkbcaptain, 2017-08-24:
I'm having the same problem atm, my thread is in the "Closed" Section.

mame82, 2017-08-25:
@jcstill 
Your kernel version is newer than the one I tested on. I've update mine and have no issues with `4.9.44+` so far.

Having to start the backdoor server manually isn't the intended behavior. If you login via SSH before the target installed the keyboard driver, it could happen that the backdoor server hasn't started already (ends up with `no screen to be detached`).
So do you see any keyboard output if you run `FireStage1` on the target ?

If yes, try to run `FireStage1 1 5000 nohide` and please report back the output of the PowerShell window.

Be sure to have installed the latest P4wnP1 release, I've fixed several minor issues

jcstill, 2017-08-25:
@Swiftb0y 
Yes it was cloned recursively.

@mame82 

> If you login via SSH before the target installed the keyboard driver, it could happen that the backdoor server hasn't started already

Sorry I even saw that you mentioned this to someone else with the same problem. That one is my fault and i should have said that I have plugged in the pi and waited ~5 min and then came back and tries SSHing. same result. I even tried changing the USB_PID so the target re-enumerated the device (and gave it at least five) which left me with the same result.

What I may end up doing is re-installing both Jessie and P4wnP1.
thanks
-J

jcstill, 2017-08-25:
@mame82 
The output of `FireStage1 1 5000 nohide` on the target is: 
`Stage1: Starting connection synchronization..`
`Stage1: Connect request sent`
`Satge1: Synced to SEQ number 10`

on my laptop i get: 
`Client didn't respond for 1 seconds`

PoSHMagiC0de, 2017-08-27:
Hello.  I have not seen this issue yet.  I am running on stretch now and did run into some issues during install that I was able to work through...after several installs.  I got smart after awhile and made a dd image of my raspian in a working state before pulling and installing.

Since you have it installed, I assume you figured out to rem out the line in the install script for the jessie check. Next, don't run it.  Instead, sudo install your python-dev, python-pip and python-pycrypto.  I then rebooted.  Do not know if I had to i just did since the pip install issues came after the apt installs.  Once I had ICS back up I sudo pip installed pycrypto and pydispatcher myself.  It seems to fail in the script.  Do not know why.  When i was done I ran the install script.  You will get red on the pip stuff again.  It is installed so no issue.  When done, everything should work even without having to add sudo to the scripts.  I had to my first time through cause something broke.  After some messing around I got things to work right with just those changes above.
I tested the HID backdoor and it works fine too.  I have not tested it against my win10 machine yet though.  I know the Powercat mod I use in my BBTPS project will open and shut down on windows 10 when fired from my agent as a process but works fine on win7.  Wonder if MS implemented another change.  Does work fine if the file is on the machine.

mame82, 2017-08-27:
@jcstill 
Weird, the initial synchronization seens to run, but the connection times out.
This would mean that the Windows process stops HID communication for more than one second, although it started without issues.

I have no glue how to reproduce this very special and weird behavior. Could it be set some kind of AV jumped in on the spawned PowerShell process, which I wasn't aware of during development?

jcstill, 2017-08-27:
@mame82 
Before I go ahead and reinstall everything from scratch, I decided to run `FireStage1 1 5000 nohide` as well as `FireStage1 2 5000 nohide` while logged in as an administrator on target. `FireStage1 1` gave me the same results, however `FireStage1 2` gave me this: 

**(sorry for the length of this)**

Both runs start with:
`Stage1: Starting connection synchronization..`
`Stage1: Connect request sent`
`Satge1: Synced to SEQ number 10`

And all the lines start with:
`Stage 1: CTRL MSG Type (number)` 

I left them in order bc I don't know whether or not order of the message #s matters

`FireStage1 2` 1st Run | `FireStage1 2` 2nd Run
------------------------ | -------------------------
285375237 | 603979776
321913856 | 285375237
755135253 | 34875906
704643072 | 1546
74388224 | 74388224
3099650 | 3099650
201326681 | 201326681
41619200 | 41619200
262663 | 262663
673251328 | 673251328
167934731 | 167934731
654311428 | 654311428
1932066816 | 1932066816
100804633 | 100804633
704805677 | 153623808
66415 | 855638026
153623808 | 710814476
855638026 | 293
710814476 | 141508864
293 | 367
33750534 | 102704239
33583121 | 272417
170400512 | 665697
272417 | 272417
0 | 
101319291 | 
665697 |

jcstill, 2017-08-28:
Like the title, I'm not sure where I went wrong in the initial install, but I reinstalled everything from scratch on a brand new SD card. In the installation, I did install these manually:
rpi-update
hostapd
dnsmasq
git
python-pip
python-dev
screen
sqlite3
inotify-tools

Seems to me that everything is working as intended. weather or not that is from me using the 2017-07-05 version of Jessie, or installing packages manually or something else, but it works now. Thank you to those of you that helped out.

closed 2017-08-28 by jcstill