RoganDawes/P4wnP1 #64: [Question] Include Mimikatz?

Anonyvius, 2017-08-24:
Is it possible to include an mimikatz ducky payload that save the passwords somewhere on the mirko SD?

mame82, 2017-08-25:
I won't include Mimikatz. But it is perfectly possible to use it in a custom payload.
Here's a workshop using metasploit on P4wnP1 https://www.madison-gurkha.com/hands-onhacking-raspberrypi-en

A similar approach could be chosen to deliver and run mimkatz.

According saving back data to SD card, this is possible to (but involves some steps).

I wrote an article for a magazine with a tutorial on exactly that, let's see if it'll get published.

Anonyvius, 2017-08-25:
Yes i trid it with my own duckyscript but i have no ideo how to save it to the SD card and load mimikatz from the SD card

Anonyvius, 2017-08-25:
Can you help me? I flashed the image from the side onto my raspberry pi zero. Ich can ssh into it bat what is the password? i tried: raspberry, raspberrypi, admin, toor, root, TheMaMe82-P4wnP1 and MaMe82-P4wnP1... so what is the password?

jcstill, 2017-08-25:
Wifi:
SSID: "P4wnP1"
Password: "MaMe82-P4wnP1"

SSH: pi@172.24.0.1
Password: "raspberry"

Anonyvius, 2017-08-25:
i know. But i used the image from the side that Mame82 linked. And with this image the raspberry does not create an wifi AP. You unterstood? I tried the way from Mame82 and it worked.

mame82, 2017-08-25:
I was more focused on the presentation provided on the link. The image given is based on a really old P4wnP1 version, which had neither WiFi support, nor a backdoor.

Anonyvius, 2017-08-25:
oh okay. Then i will use the newest version.

closed 2017-08-27 by mame82