RoganDawes/P4wnP1 #67: Help

3FF-53C-Member-DalmAsian, 2017-08-25:
Mame82 im loving your project and your code but one thing, How do i add more duckyscripts and payloads from hak5?

MJGames, 2017-08-29:
pls be more specific on the titel

PoSHMagiC0de, 2017-08-30:
I think he is trying to add more leds to his pi so he can do the light show that the BashBunny does for more led options.
Unfortunately, Mame project only uses the 1 led.  Understandable as this is the simplest way for anyone to get going without having to solder anything.  If you need more led controls, you will have to modify the project.  You may be able to get some ideas from the script he uses to trigger the led that is in his project.  You will have to go through how the GPIOs work and how to solder up leds to it and communicate and then build a routine to handle the GPIO state for the leds.  The you can modify the project to display what you want at what time.

mame82, 2017-09-01:
@PoSHMagiC0de answered the LED question better than I had done myself.

According DuckyScripts: Copy them over to the DuckyScript folder of P4wnP1 home directory to use it from hid_backdoor. The stickykeys payloads give examples on how to use DuckyScripts from this folder in a dedicated payload. The hid_keyboard2 payload gives an example how to use 'dynamic inline DuckyScript' in a payload

mame82, 2017-09-01:
Addition, if I'd like to force folks to solder, I'd prefer a hardware switch for payload selection over a status LED. Hak5 offers the toy shipping these features and they have to be paid ;-)

closed 2017-09-26 by mame82

br0k3nilluzion, 2017-09-26:
good idea.. would be cool if you had maybe 4 or 5 LED's that would show which payload you wanted to use; making it so you could have 4 or 5 payloads at the ready.. 
maybe use one of those " 3 Pin 5 Way Momentary Push Button SMD SMT Mini Tactile Switch"  type switches to click through the LED's.  would give a better feel to it.

br0k3nilluzion, 2017-09-26:
good idea.. would be cool if you had maybe 4 or 5 LED's that would show which payload you wanted to use; making it so you could have 4 or 5 payloads at the ready..  would look pretty sweet in a case.
Maybe use one of those " 3 Pin 5 Way Momentary Push Button SMD SMT Mini Tactile Switch" type switches to click through the LED's. would give a better feel to it.