RoganDawes/P4wnP1 #69: [Not an Issue] Internet from host

jcstill, 2017-08-25:
Just a quick question. Maybe I haven't done my research thorough enough but there may be someone who already knows how to do this:

Is it possible to have an Internet connection through the pi? Like is there a way that the pi can "forward" the host's connection to my laptop when I connect to the P4wnP1 wifi? I know that this would result in higher temperatures for the pi (mine functions relatively warm) because of having more things to process. It would be a cool to have a feature like this.
Like maybe it could be an option for the HID_Backdoor?

mame82, 2017-08-25:
Should be possible:
1. Connect P4wnP1 to internet through host, like shown here
https://youtu.be/QEWaIoal5qU
2. Enable IPv4 forwarding on P4wnP1
3. Enable MASQUERADING for outbound traffic on P4wnP1's internet interface (most likely `usb0`)

On the device which shold use internert:
4. Set P4wnP1's WiFi IP (172.24.0.1) as default gw
5. Set up a proper DNS (8.8.8.8 for example)

Please understand that I'm not able to give more guidance on this, as this isn't a task for P4wnp1

jcstill, 2017-08-25:
Thanks for giving me something to look into.
`Please understand that I'm not able to give more guidance on this, as this isn't a task for P4wnp1` - no worries

closed 2017-08-27 by mame82