RoganDawes/P4wnP1 #70: [Not an issue] Updating P4wnP1

GeilerGelber, 2017-08-27:
Hey I've got two questions:
How does one update P4wnP1 when a new feature has been added?
And is it possible to add own payloads by creating a new textfile in the payloads folder and a corresponding entry in the setup.cfg file?

mame82, 2017-08-27:
> How does one update P4wnP1 when a new feature has been added?

Connect P4wnP1 to Internet (for example like this https://www.youtube.com/watch?v=QEWaIoal5qU&t=5s)
CD into P4wnP1's main directory and run `git pull`. In case you want to update submodules `git submodule update --recursive`.

> And is it possible to add own payloads by creating a new textfile in the payloads folder and a corresponding entry in the setup.cfg file?

Exactly that is the intended way.

GeilerGelber, 2017-08-27:
Great thank you very much :D 
The only problem is that I tried to connect my Zero W to the internet with the help of you video but after entering the command for the route to the gateway 172.16.0.2 I still can't ping the 8.8.8.8. Have you got an idea whats wrong?﻿

mame82, 2017-08-27:
Check if 8.8.8.8 could be ping'ed from your real host (some providers block the google nameserver).

Check if P4wnP1 could ping 172.16.0.2, if not it is likely that the Windows NIC is still runing with "public" profile.

Repeat the setup of ICS, sometimes Windows messes thing up, when enabeling internet connection sharing

GeilerGelber, 2017-08-27:
Thx I already tried two times but now it worked.

closed 2017-08-27 by GeilerGelber