RoganDawes/P4wnP1 #73: issues running the hackin9 payload

andrew1999, 2017-08-27:
Hello, I'm having issues running the new hackin9 payload. I just removed the # from next to it and added one before network_only in setup.cfg. Are there any extra steps?

mame82, 2017-08-28:
Unfortunately I published a version with a path error, which is fixed now:

https://github.com/mame82/P4wnP1/commit/a022e7fbba8417270e290fe09efbcc95c8b3bd62#diff-5a404c797073f7c3a86478d7228eab0f

mame82, 2017-08-28:
Could you please test again

andrew1999, 2017-08-28:
Sure, just give me a few minutes as i reflash the sd card and p4wnp1. Unless there is a quicker way to update it.

mame82, 2017-08-28:
Connect P4wnP1 to internet (https://www.youtube.com/watch?v=QEWaIoal5qU).
SSH in.
CD into P4wnP1 main directory.
Run `git pull`.

This should work, too.

As this is a minor change, replacing the raw content of the file linke above, should also work.

andrew1999, 2017-08-28:
I've already started flashing the sd card give me 15-20 minutes and it should be done. Thank you

andrew1999, 2017-08-28:
It works perfectly now thank you!

mame82, 2017-08-28:
Nice to hear, thx for pointing me to this mistake

closed 2017-08-28 by mame82