RoganDawes/P4wnP1 #75: CreateProc crashes after so many uses.

PoSHMagiC0de, 2017-08-28:
After using P4wnP1 hid_backdoor for a bit I ran across a crash bug.  After a number of tries (it seems to be random right now) you will get the crash message from python I will list below.  You will still be in the menu system but cannot create any more processes.  Only way to resolve is to kill client, exit out of P4wnP1 and restart it with sudo.

**What I was doing**

Running Powershell processes with createproc to see what outputs will go out on standardout and can be caught by interact when done.  I was doing stuff from simple get-childitem to get services outputting stuff with out-string, convertto-csv and convertto-json.

**Error Specifics**

After a CreateProc of a set number of times it would create the process and then crash out with the error below.  The process will finish and I can interact to retrieve the results but cannot use CreateProc again unless I close out of the menu system and re-enter.

`Exception in thread P4wnP1 Server Output Loop:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "P4wnP1.py", line 240, in __output_handler
    method = pending_methods[method_id]
KeyError: 14L
`

mame82, 2017-08-28:
Although the root cause of this seems to be more or less clear to me, it is hard to build a test case on this.

Have to delay working on this, but will definitely be investigated.

According the issue: Many commands, like `CreateProcess` issue something like a remote method call (custom RPC). The python agent keeps track of the remote methods, by using a unique ID for each method in a dynamic list. This list is accessed by multiple threads and `14L` clearly isn't a valid method idea. It is likly that the real method has been deleted by another thread, trying to access the invalid method ID crashes the thread responsible for processing outbound HID traffic. The "UI thread" is still active which causes the behavior observed by you.

PoSHMagiC0de, 2017-08-29:
Ahh, so the method IDs are not only distributed by CreateProc but other functions as well that are running.  I know I was only running one proc at a time when I was testing this.  One CreateProc and then interact when it was finished.  I didn't try to hammer it with a bunch yet.  Was first working on how PS scripts would need to be formatted to use the hid_backdoor's createproc to fire off a powershell command with the script being encoded.  When I get time tomorrow I will run through the process again and see if it crashes and if I get a different method ID.  If so, that will explain it what you list is happening.  If it is the same, something might not be randomizing like it should.  I know Powershell Empire used 6-8 random character key for their jobs.  You might have to increase the ID size to prevent collisions....or check for duplicate generation and regenerate if there is one for the ID to prevent collisions if that is happening.

mame82, 2017-08-29:
Method IDs are in sequential order
https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/P4wnP1.py#L212

The error case of an overflowing 32 bit value (method ID is communicated as UInt32) isn't handled, right now. I guess the problem is based on python converting `14` to `14L` at some  point of the code. Thus i couldn't simply ignore IDs like `14L`, but need some approach to reproduce the error in order to debug and tear down the problem

PoSHMagiC0de, 2017-08-29:
This is probably going to involve adding prints a bunch of places around that functions to output what IDs they are getting...putting it around the function that hands out the ideas and/or around the function that reads them and removes them.  Probably better to output to a file so screen doesn't get cluttered and then am able to fire off freely until crash to see out.

If you want to prep some of the modules to do that to capture output into a file and have me play with it for awhile and send output file I can.  Let me know.  I like what you did here.  I am proficient in Powershell and C# .NET and getting a lot better with Python (thanks to me messing with Empire for the past couple years) but my code to serial communication and understanding is weak so me reverse engineering your code is taking some time for me to get an understanding so I can contribute more than just bug reports and feature requests.

mame82, 2017-08-29:
Putting debug prints on the relevant positions is possible. In fact there exist multiple disabled 'debug_print' command in the python classes.

Anyway, I'm developing with remote debugging for the python part of backdoor. Thus a conditional breakpoint (in case an invalid key value for method_id is used) should be more useful, to inspect what's going on.

The problem is that I have no idea on how to reliably force the error you saw.

According your cobtribution: You wouldn't have been mentioned in the installer script, if your detailed issue descriptions haven't been helpful

PoSHMagiC0de, 2017-08-30:
In the link you posted above I thought I found the issue since next() is a built in command I figured using next might have been causing issues in your function.  I was able to run more commands but after a bunch of processes I got another crash in that module with a key of 94L this time.  The function that is failing is in the handler below the function you have linked at the line I linked below.
https://github.com/mame82/P4wnP1/blob/fd54b3c44560b4ea363c66281c7b6178ae954f67/hidtools/backdoor/P4wnP1.py#L240

mame82, 2017-09-05:
Seems the error is here:
https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/Client.py#L39

Bitwise and with 0xFFFFFFFF seems to cast to the metho_id to type `long`

Using the method_id as key of a dictionary,  the value is casted into a str.

str(int(4)) != str(long(4))

Thus replacing this line https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/Client.py#L40
with `return int(next)` could maube fix this, but I still need a test case for debugging.

mame82, 2017-09-05:
Mistake in last comment, the result of this line https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/Client.py#L39 has to be casted to int.

mame82, 2017-09-05:
Hopefully fixed by this commit https://github.com/mame82/P4wnP1/commit/9ec29777ee6c0b63a4d2cd16007236ed26e56d90

mame82, 2017-09-05:
Explanation:
`4 & 0xFFFFFFFF == 4L` 
A "binary and" with a bitmask having all 32 bits set, results in a value of type `long` instead of `int`.

This means method_id is handled as value of type long on python end.

Here's the line unpacking a method_id, when it is received back from the client side agent:
https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/Client.py#L152

The id gets interpreted as `unsigned integer` which again is casted to an `int` in python.
This value is used to mark pending methods for deletion, later on. 
So there are inconsistent data types among method_ids, which again leads to invalid keys in case P4wnP1 tries to fetch a pending method for deletion using the `method_id` key.

As I wasn't able to bring up a proper test case, I can't confirm that the commit fixes the issue
and hope for feedback on this.

If no further errors occur, I'm goning to close the issue with some days delay.

PoSHMagiC0de, 2017-09-06:
I will give it a spin today if work dies down else I will definitely run it through tests tonight and let you know.

PoSHMagiC0de, 2017-09-07:
@mame82 
Crash still happening.  New issues have cropped up too.  Crashed with a key error of 22 this time pretty dang quickly.  GetClientProcs return an error about not finding a procedure.  I am at work when this happened so will get the exact function it cannot find.  After proc error through, interacting freezes all of P4wnP1.  Used to work and then not let you create new procs.

mame82, 2017-09-07:
> GetClientProcs

The remote part of this method was removed for redesign, but I forgot to remove the command from the server shell

mame82, 2017-09-07:
Next attempt:

https://github.com/mame82/P4wnP1/commit/8c3c279846ff4c6e6a475aa7d29c0f8bf81344e6

For whatever reason I wasn't able to reproduce the issue in a debug session, but did some analysis:

Pending client methods are deleted here https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/Client.py#L185
... as the comments states, I hade concerns according thread safety while implementing it this way.

This concerns were caused by the fact, that the output processing for pending methods from here
https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/P4wnP1.py#L258
... is handled of by a different thread, which isn't aware of removed methods, while iterating over the pending ones.

As removed methods should have finished processing, anyway, I have added in a `try ... except ..` which essentially results in output processing ignoring methods which aren't present anymore.

Again, this is hard to reproduce manually (method has to be deleted by input thread, while output thread is iterating over the methods), but I guess it is fixed now

PoSHMagiC0de, 2017-09-07:
I was thinking the same thing, that a race condition was happening.  I been prowling through your code to see where locks could be implemented.  Hard to find as I am used to inheriting threading into my classes so I can implement locks internally.  You are passing handler functions to the thread constructor to create threads so do not know how to handle locks in this situation.  I assume you can use them in your handler methods since they turned into threads at runtime.  I will do a pull tonight to update my install and try it again.

PoSHMagiC0de, 2017-09-08:
So I got the latest pull and have been hammering my Win10 machine with CreateProc and interact to get results for about an hour now while watchhing disJointed on Netflix.  It has not even studdered.  No issues seen so far.  I will continue to test to see how it works.

mame82, 2017-09-08:
Sounds promising.

According a "Monitor Lock":

I avoided it in this case, because I'd have to lock 'pending methods' during the whole iteration of output processing (which could take relatovly long) only to allow deletion of a single method during input processing. This would pause the input thread too long.

Blocking the method access itself wouldn't help, as the error affects the data structure holding the methods.

The proper solution would be to create a second data structure, collecting methods which are marked for deletion and delete them in the output thread. But the current solution should work, too, as the exception should only trigger once per method. In the next output iteration the method is already gone.

The cause for having these things a bit messy, is that my current IDE isn't able to break execution of a single thread while debugging, but pauses all threads. So it gets a bit hard debugging inter thread communication

mame82, 2017-09-19:
Seems to be solved, no issues so far

closed 2017-09-19 by mame82

PoSHMagiC0de, 2017-09-19:
Yeah.  I meant to get back to you on this one.  I saw it there 2 days ago
still and began to run again through some tests just to be sure at work but
was pulled away with the cries of many customers who were using their
machines the wrong way at the same time but swear something that never
worked, worked.  :-P

commit 8c3c2798 (2017-09-07): Erros case handling, should fix #75