RoganDawes/P4wnP1 #85: Download tools on P4wnP1

Hugand, 2017-09-01:
Hi everyone! So, I recently installed P4wnP1 on my RPi0w and I need to download some tools to the pi, but since it uses his own network and can't use the internet normally I can't get them. How can I solve this??

mame82, 2017-09-01:
Happy to have a new user, help is on the way...
Here: https://youtu.be/QEWaIoal5qU

Hugand, 2017-09-01:
Thank you so much! Is it possible to use the RPi as HID and at the same time as a mass storage device?
Nice job on the P4wnP1 btw!

mame82, 2017-09-01:
Yes, hakin9_tutorial payload is an example on this

Hugand, 2017-09-01:
Ok, thanks a lot!

Hugand, 2017-09-01:
Oh, one more thing, how can I run a ducky payload instead of the default payloads like /payload/hid_keyboard.txt??

mame82, 2017-09-01:
Same answer, hakin9_tutorial shows how to use a DuckyScript payload

Hugand, 2017-09-01:
ok, thanks once again

closed 2017-09-05 by mame82