RoganDawes/P4wnP1 #86: Is the setup process easy with a Pi Zero W?

thebeastgamer25, 2017-09-01:
Is all I need to buy a Pi Zero W and a micro sd card, to get everything setup easy? How would I setup WiFi, since it is a bit tedious on the normal Pi Zero.

mame82, 2017-09-01:
If you're able to do a headless installation like described in INSTALL.md you're good to go (assuming your PC has a SD card reader and you have an Micro-USB cable to connect to your target ).

I would recommend to get an USB OTG and HDMI adapter, too. In case you boot a payload which doesn't provide network access this could be used to boot back into interactive mode (attaching keyboard and monitor). If you don't like to use it like this, you could still change the setup directly on the SD card, but you need an OS capable of handling `ext4` partitions.

thebeastgamer25, 2017-09-01:
@mame82 does widnows support that?

mame82, 2017-09-01:
> does widnows support that?

If you refer to headless installation, short answer is yes.
More precise answer is: You have to manually change `inf` files, because raspbian's default Ethernet USB setup is detected as Serial COM port, instead of RNDIS. So I don't give instruction on this, as the process is far more complicated compared to Linux.

Once P4wnP1 is installed, it could be used with Windows without any problems.

So I suggest to grab a bootable copy of Kali and follow instruction given in INSTALL.md or follow the hints given by @Seytonic on youtube, which needs adfitional hardware.

thebeastgamer25, 2017-09-01:
@mame82 how do i disable the numlock trigger payload to clean my tracks?

mame82, 2017-09-01:
> clean my tracks?

Could you get more precise?

thebeastgamer25, 2017-09-01:
@mame82 make it so the victim cant activate the payload by pressing numlock after the pi has been removed?

mame82, 2017-09-01:
> make it so the victim cant activate the payload by pressing numlock after the pi has been removed?

Still not sure what you mean, of course the payliad doesn't trigger if P4wnP1 is removed

thebeastgamer25, 2017-09-01:
@mame82 oh, then whats the remove payload for?

mame82, 2017-09-01:
Oh, you're talking about the LockScreen backdoor based on Sticky Keys. I suggest watching this and the purpose of the removal part becomes clear:

https://youtu.be/uu15fIb9vbs

thebeastgamer25, 2017-09-01:
@mame82 should probaly update desc to say its published, btw

thebeastgamer25, 2017-09-01:
@mame82 also, whats ur fav payload?

mame82, 2017-09-01:
All of these payloads are Proof of Concepts to give real pentester an quick introduction on bow to deploy well known techniques to the P4wnp1 framework. To be honest, P4wnP1 isn't aimed at 'Script Kiddies' searching a ready-to-fly solution. It's meant to be used by redteamers and pentesters, who know how to modify payloads or develope own ones, fitying the needs of an engagement.

Anyway, there's an exception: The hid_backdoor is fairly advanced and I only know two tools able to do something similar, both are based on custom build hardware and have been presented at conferences.

If you want to get a brief insight I suggest watching https://youtu.be/HTrQHZtNyBw

mame82, 2017-09-01:
>  should probaly update desc to say its published, btw
Thanks for the hint

thebeastgamer25, 2017-09-01:
@mame82 yeah, i wasnt gonna use it like a script kiddie, probaly gonna make an all-in-one payload or something. after the setup is done with the kali box, can i just use it on windows? lastly, do u think its worth it for a tech enthusiast like me who just want to play around with it?

thanks

mame82, 2017-09-01:
Let me answer like this: A tech enthusiast is able to bring the device up. A pentester is able to understand how things work, at which points it gets dangerous and what could go wrong.

I don't take responsibility if somebody uses P4wnP1 the wrong way and of course it mustn't be used against foreign system without authorization of the owner.

This is why for payloads relying on security issues of external software, got reported to the vendor.

So yes, if you want to test it against your own systems in order to make them more secure, you should give it a try.

On a really hardened setup not a single one of the payloads shipped from the repo should succeed.

Unfortunately not every system is hardened properly and P4wnP1 is one of many tools to test for this.

thebeastgamer25, 2017-09-01:
@mame82 last thing, would i have to do the WiFi setup every time i go to a new location?

mame82, 2017-09-01:
No, P4wnP1 doesn't run a WiFi client, but provides an Access Point (location independent)

closed 2017-09-18 by mame82