RoganDawes/P4wnP1 #88: Windows Lock Picker Hashing Issue?

Wobble107, 2017-09-02:
Hello,

So I decided to try out the Windows Lock Picker payload but I've been having some difficulties, I've tested the payload on two machines. The first test ended in receiving a 'hash' file that only contained the username of the computer and a 'cracked hash' file that was blank. The second doesn't change from two flashes - waiting for the hash, even after a few hours. Any idea why this is happening? 

NB: Quite happy that my PC's aren't getting their hashes stolen.

Thanks.

mame82, 2017-09-02:
> Quite happy that my PC's aren't getting their hashes stolen.

In fact you are one of the few with the intended behavior. This payload shouldn't work at all, unfortunately it does most of the time.

I'm afraid I'm not able to give more details right now. I'll revisit the issue when I am allowed to.

Anyway, I'd be happy if you could give some details on your target (exact OS version + installed 3rd party security software, like AV, if applicable)

TheKoopaGaming, 2017-09-05:
I'm also having the same problem- I'm running Windows 10 Build 1607 and have Avast and Malwarebytes installed, but I tested it with both of them disabled and still had the same problem.

Thanks for any help!

Wobble107, 2017-09-18:
Hi there,
Apologies for the late reply.

The first machine I am using is an XPS15 (laptop) running Windows 10 Home, Version: 1703, OS Build: 15063.608, AV: Kaspersky Total Security.

The second machine: Custom Desktop, Windows 10 Home, Version: 1703, OS Build: 15063.540, AV: Kaspersky Total Security.

I have tested this on two other machines however, I cannot access them at this time and will update this later. Hope this helps let me know if there is any more information I can provide.

h4ckzard, 2017-10-05:
Hi there. 
Today i'm tested P4WnP1 a lot of times and noticed that the device does not extract the hash, if the account belongs to Microsoft (ex. login@gmail.com). Continuous flashing twice (blink 2). 
But after all, offline login to the account is possible. Then the question is: where does he store NTLM?
Also, the hash was not extracted, if Win 10 authentication uses Pin-code authorization.
Hm...

mame82, 2017-10-12:
The inner workings of the attack are described here https://github.com/mame82/P4wnP1/blob/master/writeup_lockpicker.md
and have been addressed with KB4041691... no further support on this, sorry

closed 2017-10-12 by mame82