RoganDawes/P4wnP1 #89: where are the downloaded files saved?

Anonyvius, 2017-09-02:
With the backdoor payload i downloaded an .jpg file from my download directory. But, where is the downloadet .jpg image saved? I can not find it.

Anonyvius, 2017-09-02:
oh. i found it. But now. how can i download it from the raspberry to my latop?

mame82, 2017-09-02:
I haven't managed to document the filesystem functions of the backdoor. If you ever used meterpreter it should be easy.

- `pwd` shows current remote directory
- `cd` changes remote directory
- `ls` lists a remote directory

All three command exist for local file browsing, too, if an `l` is prepended (`lls`, `lpwd`, `lcd`).

Now to answer your question:

You have two working "current directories" - the one on P4wnP1 and the one on thr remote machine. If you use `download` without absolute target path, the current working dir of P4wnP1 us used as target. By default this should be at `P4wnP1/hidtools/backdoor`.

You could dowload files to a different directory using `download c:\path\to\sourcefile.jpg /path/to/download/targetfile.jpg`.

The same would be achieved using:
`lcd /path/to/dowload`
`download c:\path\to\sourcefile.jpg targetfile.jpg`

Or if you want to keep the filename:
`lcd /path/to/dowload`
`cd c:\path\to`
`download sourcefile.jpg`

mame82, 2017-09-02:
Sorry, your question was how to transfer from Pi to a PC.

Solution 1)
Use the HID backdoor's `upload` command "against" your Laptop

Solution 2)
Use the Network-Only payload and transfer files using SCP or WinSCP. 

Solution 3)
Mount the SD card to a system capable of reading ext partitions (f.e. Linux)

Anonyvius, 2017-09-05:
I have now done it with WinSCP but i did not used the "network only" playload it worked with the hidbackdoor payload too.

mame82, 2017-09-05:
Yes, my fault... SSH/SCP to the HID_backdoor is perfectly possible. Constantly working on this sometimes prevents me from seeing the wood for the trees

jcstill, 2017-09-06:
What about using apache2 on pi?

mame82, 2017-09-06:
> What about using apache2 on pi?

Many people are doing this. If you're going to use apache2 for HTTP based file delivery only, I suggest using something more lightweight (tinyHTTP os python SimpleHTTPServer).
If you want to get it bidirectional: yes, use a webserver with webdav support.

jcstill, 2017-09-06:
Thanks for the suggestion. Would this be a feature that might be added?

mame82, 2017-09-06:
Not sure why I should add this.

Run `sudo apt-get install apache2` while connected to internet.
Everything is there, Raspbian is a full fledged Linux distro.

PoSHMagiC0de, 2017-09-09:
Another option is to install python-impacket and use the smbserver.py script to fire up a quick SMB server pointing to whatever subfolder you want.  You just have to make sure to have networking ability on.  RNDIS is good at this as it seems to give out an IP for both nix and windows systems.

@mame82 
Is there anyway to crank down the interface reported network speed?  I believe in your install you patched it with the patch that cranks the speed up.  If you are backdooring, you do not want to interrupt the victims traffic but I noticed if I have network enabled for RNDIS it will take over my traffic.
I think I mentioned in another thread that I noticed RNDIS assigns an IP just fine for all systems but cannot use it for ICS (can use it for SSH though).  ECM enabled it will not autoassign the host an IP so I have to manually add it to my interface it brings up but I can ICS.  I stopped bringing both up because I noticed the reason my ICS didn't work in network only mode is because RNDIS high speed network has taken over my traffic hehe.

closed 2017-09-26 by mame82