RoganDawes/P4wnP1 #90: No screen to be attached

zezto, 2017-09-04:
Im trying to use hid_backdoor but when i boot it up with the payload it says no screen to be attached. Just little background i also connected it to the internet following mame82 tutorial. Any ideas what going on?

asing012, 2017-09-04:
This also happened with me, I would advise you to change your cable.

gloglas, 2017-09-04:
Me too after update to last version. Double check installation process and you internet speed + cabel.

mame82, 2017-09-04:
> Me too after update to last version

Can anybody confirm, that this was caused by an update? I'm not able to reproduce the issue

teslatickles, 2017-09-04:
I apologize if this suggestion is offensive (I'm new to programming, in general), but, just in case:

I tried doing the backdoor payload on a MacBook (target) using my smartphone to no avail. I finally loaded up parallels desktop to run windows 10, and successfully gained entry (the P4wn shell pops right up) to my Windows setup. I felt stupid for trying so hard to get it to work on Mac (as the target).

gloglas, 2017-09-04:
and also i have bad driver recognition on win7 even on new machine.

PoSHMagiC0de, 2017-09-05:
I will have to load up a new SD card and give the installation and once through to see if there are any issues when I get home.
As far as the Macbook as a target, the HID_Backdoor only support Windows as a target for now.  Maybe in a future update mame will add support for *nix systems, maybe using a python agent.  For not you can only do rubber ducky like commands for *nix systems, this includes OSX.

Anonyvius, 2017-09-05:
My tip. try to use another mikro USB cable. I had this error too. but then i took another cable and it worked.

PoSHMagiC0de, 2017-09-09:
Okay, after lots of redos I have not been able to reproduce error.
I have seen this on my very first install on stretch but that has since been resolved.
If you have not already, I would redo the pi and reinstall P4wnP1 as the updates mame has made to the installer should resolve this issue.

TeCoEd, 2017-09-12:
I now have the same issue,  on older version and latest version.  Original card etc worked last week but has now stopped.

jcstill, 2017-09-12:
@TeCoEd are you SSHing into the pi? 

This may also solve some other people's problems:

If so, try waiting longer before you SSH to allow the pi to get everything started up and running. As stated [here](https://github.com/mame82/P4wnP1/issues/62) by mame. I found that in my case it seems to work after a couple of minutes from when I plug the pi into target. (dont ssh as soon as you connect to wifi, give it some time)

Make sure you are using a cable that has data lines. If you just use a charging cable it wont work.
Post the output of `uname -a` as well so we can see what version raspbian you are running.

If these things still bring no result, you can try:
`sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py` to see if it starts at all.

or

Reinstall everything from scratch, new raspbian image and everything. If you want, you can install these packages manually:

apt-get update + upgrade
rpi-update
hostapd
dnsmasq
git
python-pip
python-dev
screen
sqlite3
inotify-tools
(manually installing worked for me when I was having the issue, but I don't bother with it now as install.sh does it for you)

Hope this helps
-J

TeCoEd, 2017-09-13:
Thanks for the hints, I get this error now -bash /tmp/blink_count: permission denied, but, I have made no changes to the Pi or config.

TeCoEd, 2017-09-13:
I ran the sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py code and got this error

Traceback (most recent call last):
  File "/home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py", line 1248, in <module>
    HIDin_file = open(dev_file_in_path, "rb")
IOError: [Errno 2] No such file or directory: '/dev/hidg1'
Cleaning Up...
Traceback (most recent call last):
  File "/home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py", line 1284, in <module>
    ll.stop() # send stop event to read and write loop of link layer
NameError: name 'll' is not defined

jcstill, 2017-09-13:
What is the output of 'uname - a'?

>I have made no changes to the Pi or config.

So you are doing the exact same things as before? And you get a different error?

mame82, 2017-09-13:
@TeCoEd
The `/tmp/blink_count` error is a known bug and I opened issue #93 to keep track pf this.

The error stating that `/dev/hidg1` is missing is the real root cause of your problem. This happens because P4wnP1 isn't able to bring up the HID devices, which is most likely caused by a kernel with broken HID gadget driver. To be sure about this assumption the `uname -a` output is needed, as @jcstill already stated.

If you're on a wrong kernel version (latest raspbian shipped with broken kernel module), I suggest a clean re-install, closely following the instructions given in INSTALL.md

TeCoEd, 2017-09-14:
output is this?

Linux MAME82-P4WNP1 4.9.48+ #1034 Fri Sep 8 13:55:13 BST 2017 armv6l GNU/Linux

I see that there is a September OS update for Stretch

Thanks for your help so far

mame82, 2017-09-14:
Weird, this is a newer kernel than the one I tested.
It was working on `4.9.45+`

Have to do some tests...

mame82, 2017-09-14:
Updated to latest kernel `4.9.50+`, HID backdoor works without issues. So I suggest a clean install from scratch

mame82, 2017-09-14:
One more thing...
Could it be possible that you haven't chosen the `hid_backdoor` payload ? In this case the HID device used for covert channel communication wouldn't be loaded and thus `/dev/hidg1` wouldn't exist !

mame82, 2017-09-18:
Is the problem solved ?

TeCoEd, 2017-09-18:
Yes all solved this afternoon.

I rebuilt from scratch on a new SD card and it works perfectly.  Thanks for all your help and keep up the good work on a great project.

closed 2017-09-18 by mame82